reinhard-brandstaedter/solarflow-control #266: Behave from hub2000 after power failure

joe63, 2024-06-18:
The sf-control container doesn't restart After powerfailure. I Was called from my wife while i'm sitzing in the train. She said that we had a power-failure. Now everything runs , but not the sf-control container.
Is it a problem for the hub 2000, when i Restart the  container at late afternoon?

welli2009, 2024-06-18:
It shouldn’t be a problem. When you’re running the container on a Linux based system, I would recommend to add a cron job for the future

reinhard-brandstaedter, 2024-06-18:
How you start/restart your services is out of the scope of this project. You might want to look into various strategies: https://docs.docker.com/config/containers/start-containers-automatically/

For the SF Hub it doesn't really matter if sf-control is restarted, there are only a few persistent topics held in MQTT (last battery full/empty timestamps, charge through mode, ...) so you might want to have some persistence for your MQTT broker as well in case that goes down.

joe63, 2024-06-18:
_How you start/restart your services is out of the scope of this project. You might want to look into various strategies: https://docs.docker.com/config/containers/start-containers-automatically/_

Thank you (again) !!
Now I changed it to restart-always.

closed 2024-06-18 by reinhard-brandstaedter