reinhard-brandstaedter/solarflow-control #272: Support for Zendure Hyper 2000

maxp1256, 2024-07-03:
Hello!

are there any plans for supporting the new Hyper 2000? I've an Shelly 3EM, a photovoltaik without an hybrid inverter and the awattar.at hourly tariff. I would expect to be able to use the cheap hours in the night (2-5) to load the batteries, skip the rise in the morning hours (6-9) with zero net usage, continue reloading&flatting burst peaks (washing mashine/dishwasher heating or cooking) during the day with pv and skip the next high during 18-21 until batteries are empty.

I don't expect that it's possible to cover this with the standard app so I would be happy to have a very simple possibility to set mode (charge/discharge) and wattage locally, fast, reliable and without cloud/internet demand.

reinhard-brandstaedter, 2024-07-04:
Servus,
I have no other device available than my own SFHub 1.2k. So I can’t tell you exactly. You might be able to disconnect the Hyper from Zendure Cloud the same way as the SF hubs with the sf-bt-manager and then you would get all data reported to a local MQTT broker, which is also used to relay commands to the Hyper2000 (given it works similar to the other SF products).
It could be though that Zendure has implemented secure MQTT by now which could be an issue.

The shelly integration is likely a data exchange on the zendure backend between their mqtt and shelly cloud. But even so, if your hub is using a local mqtt you likely can also steer it via simulated shelly reporting (smart matching mode)

ralf-koe, 2024-07-09:
Does that
> you likely can also steer it via simulated shelly reporting (smart matching mode)
mean, that I can write Shelly data via MQTT?

maxp1256, 2024-07-09:
Hello,

a local mqtt service/proxy which modifies the real shelly em data sounds very promising. Do you know how it's working currently? (the marketing documents from zendure are talking about a link between shelly and zendure cloud) but I don't think that this would work because of instable internet connectivity and WAN delays.
Probably the "linking" is only the interchange of the local mqtt endpoint (IP+Port)?

reinhard-brandstaedter, 2024-07-10:
No, the original Zendure/Shelly integration can only happen via the public clouds. So both your devices (shelly & solarflow) have to be online.
For users like me (shellies are only reporting locally) this wouldn't work.
The hub has a input property ```smartPower``` which I assume is used to provide the readings of a Shelly (or any other power meter device) which then is used to adjust the outputLimit. Also the ```smartMode``` will control in which. mode (smart matching) the hub is working.
[image]
So if you want to try your approach you'll need to set the smartMode accordingly (check values by setting it from the App) and then publish data via the ```smartPower``` property.
If I'm correct the hub will then adjust it's ```outputLimit```.
There are for sure some ramifications, like how often that limit can be updated (the hub was never the fastest in switching), ..

Raudi1, 2024-08-24:
Today I finally had some time to try solarflow-control on my Hyper 2000. Taking it offline worked, but I couldn't really get it to work completely. That's expected, since the device is different from the hub, but I'd really like to have local control of my Hyper 2000. I'm sadly no developer and can only do a bit of coding, so I'd only be able to help.
Ideally this software would control the integrated inverter to charge the Hyper2000 with excess energy and to discharge it when there's power taken from the utility. I'd be happy with being able to set the charge/discharge power via MQTT though. Then I could let EVCC or Home Assistant handle it.
After taking the Hyper2000 offline, there's no MQTT data aside from this:
[image]
Here's what the log shows
```
2024-08-24 13:59:55,579:INFO: MQTT Host: 192.168.1.1:1883
2024-08-24 13:59:55,579:INFO: MQTT User: MQTT/secret
2024-08-24 13:59:55,579:INFO: Solarflow Hub: ja72U0ha/xxxxxxx
2024-08-24 13:59:55,579:INFO: Limit via inverter: False
2024-08-24 13:59:55,579:INFO: Control Parameters:
2024-08-24 13:59:55,579:INFO:   MIN_CHARGE_POWER = 10
2024-08-24 13:59:55,579:INFO:   MAX_DISCHARGE_LEVEL = 800
2024-08-24 13:59:55,579:INFO:   MAX_INVERTER_LIMIT = 800
2024-08-24 13:59:55,579:INFO:   MAX_INVERTER_INPUT = 1190
2024-08-24 13:59:55,579:INFO:   SUNRISE_OFFSET = 60
2024-08-24 13:59:55,579:INFO:   SUNSET_OFFSET = 60
2024-08-24 13:59:55,579:INFO:   DISCHARGE_DURING_DAYTIME = True
2024-08-24 13:59:55,691:INFO: IP Address: xxx.xxx.xxx.xxx
2024-08-24 13:59:55,691:INFO: Location: Secret, North Rhine-Westphalia, Germany
2024-08-24 13:59:55,691:INFO: Coordinates: (Lat: xx, Lng: xx)
2024-08-24 13:59:55,693:INFO: No config setting found for option "control_bypass" in section solarflow!
2024-08-24 13:59:55,694:INFO: Publishing Homeassistant templates...
2024-08-24 13:59:55,732:INFO: Published 33 Homeassistant templates.
2024-08-24 13:59:55,732:INFO: Triggering telemetry update: iot/ja72U0ha/xxxx/properties/read
2024-08-24 13:59:55,732:INFO: Using OpenDTU: Base topic: solar/xxxx, Limit topic: solar/xxxx/cmd/limit_nonpersistent_absolute, SF Channels: [3], AC Limit: 1200
2024-08-24 13:59:55,732:INFO: No config setting found for option "scaling_factor" in section smartmeter!
2024-08-24 13:59:55,732:INFO: Using Smartmeter: Base topic: shellypro3em/status/em:0, Current power accessor: total_act_power, Total power accessor: total_aprt_power, Rapid change diff: 500W, Zero offset: 20W, Scaling factor: 1
2024-08-24 13:59:55,732:INFO: Connected to MQTT Broker!
2024-08-24 13:59:55,733:INFO: Hub subscribing: /ja72U0ha/xxxx/properties/report
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/solarInputPower
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/electricLevel
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/outputPackPower
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/packInputPower
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/outputHomePower
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/outputLimit
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/inverseMaxPower
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/masterSoftVersion
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/pass
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/passMode
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/batteries/+/socLevel
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/telemetry/batteries/+/totalVol
2024-08-24 13:59:55,733:INFO: Hub subscribing: solarflow-hub/xxxx/control/#
2024-08-24 13:59:55,733:INFO: Turning hub buzzer OFF
2024-08-24 13:59:55,733:INFO: DTU subscribing: solar/xxxx/0/powerdc
2024-08-24 13:59:55,733:INFO: DTU subscribing: solar/xxxx/0/efficiency
2024-08-24 13:59:55,734:INFO: DTU subscribing: solar/xxxx/+/power
2024-08-24 13:59:55,734:INFO: DTU subscribing: solar/xxxx/status/producing
2024-08-24 13:59:55,734:INFO: DTU subscribing: solar/xxxx/status/reachable
2024-08-24 13:59:55,734:INFO: DTU subscribing: solar/xxxx/status/limit_absolute
2024-08-24 13:59:55,734:INFO: DTU subscribing: solar/xxxx/status/limit_relative
2024-08-24 13:59:55,734:INFO: DTU subscribing: solarflow-hub/+/control/dryRun
2024-08-24 13:59:55,734:INFO: Smartmeter subscribing: shellypro3em/status/em:0
2024-08-24 13:59:55,777:INFO: Taking over bypass control: False
2024-08-24 13:59:55,778:INFO: Updating SUNRISE_OFFSET to 60 minutes
2024-08-24 13:59:55,778:INFO: Updating SUNSET_OFFSET to 60 minutes
2024-08-24 13:59:55,778:INFO: Updating MIN_CHARGE_POWER to 10 W
2024-08-24 13:59:55,778:INFO: Updating MAX_DISCHARGE_POWER to 1200 W
2024-08-24 13:59:55,778:INFO: Updating DISCHARGE_DURING_DAYTIME to True
2024-08-24 13:59:57,964:INFO: Rapid rise in demand detected, clearing buffer!
2024-08-24 13:59:57,966:INFO: HUB: S:-1.0W [ ], B: -1% (-1), V:-1.0V (-1.0), C: 0W, P:False (manual, possible), F:-1.0h, E:-1.0h, H: -1W, L: -1W
2024-08-24 13:59:57,966:INFO: INV: AC:800W, DC:1150.8W (285.2|304.9|290.8|269.9), L:800W (500.0W/channel) [ -1W]
2024-08-24 13:59:57,966:INFO: SMT: T:Smartmeter P:3304.0W [ 3304.0,3304.0,3304.0 ]
2024-08-24 13:59:57,967:INFO: SMT triggers limit function: 3304.0 -> 3304.0: executed
2024-08-24 14:00:14,469:INFO: Determined inverter's max capacity: 2000.0
```
I

welli2009, 2024-08-24:
Hi Vrabetz,
   
  
   Your logs do not contain the necessary informations needed to develop your control. Try using MQTT Explorer to gather the necessary informations and attach screenshots from MQTT Explorer with the topics.
   
  
   welli

Raudi1, 2024-08-24:
> Hi Vrabetz, Your logs do not contain the necessary informations needed to develop your control. Try using MQTT Explorer to gather the necessary informations and attach screenshots from MQTT Explorer with the topics. welli

I'll try, but currently I can't connect to the Zendure MQTT. Was working earlier this day.
Apparently the Hyper2000 uses the same data as the Hub and extends it: https://github.com/Zendure/developer-device-data-report?tab=readme-ov-file#the-fields-extended-by--hyper2000--ace-1500-based-on-solarflow
The topic is always the same `ja72U0ha/"DeviceID"/state` and the JSON only contains part of the values. Here's a few examples:
```
'{"hubState":0,"remainInputTime":59940,"packState":2,"sn":"xxxxx","remainOutTime":281}'
'{"heatState":0,"sn":"xxxx"}'
'{"hyperTmp":3131,"sn":"xxxx"}'
'{"inverseMaxPower":600,"inputLimit":116,"outputLimit":30,"acMode":1,"sn":"xxxx"}' #acMode 1 is input from the grid; 2 is output
'{"outputHomePower":261,"packInputPower":270,"electricLevel":75,"sn":"xxxx"}'
'{"hyperTmp":3141,"sn":"xxxx"}'
```

linuxstony, 2024-09-04:
> Hi Vrabetz, Your logs do not contain the necessary informations needed to develop your control. Try using MQTT Explorer to gather the necessary informations and attach screenshots from MQTT Explorer with the topics. welli

Which information would be important particularly? Currently i have the Hyper connected to the official global MQTT  Server which is bridged to my local one.  I would be as willing to assist as @Raudi1 is :)

jschroeter, 2024-09-16:
> I'd be happy with being able to set the charge/discharge power via MQTT though

This is also what I'm trying as a first step but I'm running into the same issue. When Hyper is locally connected to my HA MQTT (worked with [this](https://github.com/reinhard-brandstaedter/solarflow-control/issues/204#issuecomment-2351493677)) I only receive the `register/replay` message, nothing else. I tried publishing via `iot/${productKey}/${deviceKey}/properties/read` or `iot/${productKey}/${deviceKey}/properties/write` topics but so far nothing happened.
@reinhard-brandstaedter could you maybe explain the local MQTT API a bit further? Which topics to we need to publish to receive more values or to change values? Thanks a lot for all your efforts!

Basically I'd like to have a local smart control for Hyper (AC-only, no panels connected), via Shelly 3EM and HA. The native cloud-based Shelly integration works in general, but it's a bit slow and I would prefer it to work 100% local.

jschroeter, 2024-09-16:
Maybe also helpful as a reference: https://github.com/nograx/ioBroker.zendure-solarflow
From the code it seems like they support controlling the Hyper - but via the cloud API.

reinhard-brandstaedter, 2024-09-23:
@jschroeter once the hub is registered to the local MQTT it should start sending data to a topic like this:
```
/${productKey}/${deviceKey}/properties/report
```
(note the leading "/" which is usually not a standard/valid MQTT topic as it results in an empty top node).
There were cases reported where the hubs aren't reporting immediately after the register switch. (I personally had this once for no obvious reason).
Sometimes a few register retries or a few reboots of the hub helped - with some waiting in between.
I always tried this:
- register the hub to the own MQTT (make sure your MQTT allows anonymous logins, or ensure you have the correct user/pwd added to MQTT, with the hardcoded user and password)
- wait and see if data arrives, if not after 15min restart the hub
- wait again, maybe disconnect the hub from WiFi (force WiFi reconnect from your AP/Router)
- if that doesn't help, register with the mobile App to zendure, try again.

Once the hub reports via the above topic, solarflow-control will read those and "pretty-print" them to
```
solarflow-hub/${productKey}/telemetry
```

reinhard-brandstaedter, 2024-09-23:
@jschroeter you also might find this useful, i summarized the properties and how the hubs can be controlled directly:
https://github.com/reinhard-brandstaedter/solarflow-control/tree/dev/doc

MaisonGF, 2024-10-27:
@reinhard-brandstaedter not "smartMode" or neither "smartPower" reported here by the Hyper disconnected :

[image]

And not in control : 

[image]

Any idea ? Did i miss a parameter ?

reinhard-brandstaedter, 2024-10-28:
@MaisonGF we are just unwrapping whatever is reported by the Zendure device via it's native .../properties/report topic.
There is no logic in sf-control to extract anything device specific. It just takes what is coming in and creates a separate topic for it. If you don't see any properties then they are either not sent or not yet sent...

cmuster, 2024-11-30:
@jschroeter I'm running into the same issue here with my Hyper2000. Did you figure it out in the meantime?

When just connecting via BT I'm getting all the properties, but when connecting to my local MQTT broker 
I'm just receiving 2 topics in MQTT explorer (not sure what id gDa3tb is in the 2nd topic):
```
iot/ja72U0ha/<deviceId>/register/replay

/gDa3tb/<deviceId>/time-sync
```

I added a user to my MQTT broker with name <deviceId> and pw generated by [this snipped](https://github.com/reinhard-brandstaedter/solarflow-control/issues/204#issuecomment-2351493677) from @reinhard-brandstaedter

jschroeter, 2024-12-01:
@cmuster unfortunately not yet. Only tried it one more time but without success and didn't invest more time yet. But I'm still interested and want to try it again soon.

reinhard-brandstaedter, 2024-12-02:
@cmuster can you post a full log of bt-manager when just connecting via bluetooth (e.g. via the -i option only).

cmuster, 2024-12-07:
@reinhard-brandstaedter sorry for the late reply, finally had time to give this another try. 
btw: when I try to reconnect back to the Zendure cloud via BT manager script this also fails - have to go through the app in this case. I did push the power button 3 seconds to reset network settings before running the disconnect/connect script, as is the process when doing it via app. Also tried restarting the Hyper afterwards.

Here's the info log via bluetooth:

```
2024-12-07 10:08:01,694:INFO: scan for: zene
2024-12-07 10:08:02,689:INFO: Found device: <mac> ZenE1_38
2024-12-07 10:08:04,396:INFO: Services:
2024-12-07 10:08:04,397:INFO: 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
2024-12-07 10:08:04,398:INFO: 00001800-0000-1000-8000-00805f9b34fb (Handle: 5): Generic Access Profile
2024-12-07 10:08:04,398:INFO: 0000a002-0000-1000-8000-00805f9b34fb (Handle: 12): Vendor specific
2024-12-07 10:08:04,399:INFO: 00112233-4455-6677-8899-aabbccddeeff (Handle: 18): Unknown
2024-12-07 10:08:04,595:INFO: {'method': 'read_reply', 'deviceId': '<deviceId>', 'success': 1, 'properties': {'getAll': 1}}
2024-12-07 10:08:04,833:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'packNum': 2}, 'packData': [{'sn': '<serialN>'}, {'sn': '<serialN>'}]}
2024-12-07 10:08:04,933:INFO: {'messageId': 799606, 'method': 'error', 'deviceId': '<deviceId>', 'timestamp': 1733562476, 'offData': 1, 'data': []}
2024-12-07 10:08:05,073:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'masterSwitch': 1, 'electricLevel': 49, 'buzzerSwitch': 0}}
2024-12-07 10:08:05,355:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'socSet': 960, 'solarInputPower': 0, 'packInputPower': 0}}
2024-12-07 10:08:05,514:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'outputPackPower': 0, 'outputHomePower': 0, 'gridInputPower': 0}}
2024-12-07 10:08:05,693:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'solarPower1': 0, 'solarPower2': 0, 'remainOutTime': 59940}}
2024-12-07 10:08:05,812:INFO: {'deviceId': '<deviceId>', 'method': 'BLESPP'}
2024-12-07 10:08:05,893:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'remainInputTime': 59940, 'packState': 0, 'hubState': 0}}
2024-12-07 10:08:06,173:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'masterSoftVersion': 8465, 'masterhaerVersion': 0, 'inputMode': 0}}
2024-12-07 10:08:06,333:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'blueOta': 1, 'pvBrand': 0, 'VoltWakeup': 0}}
2024-12-07 10:08:06,513:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'minSoc': 150, 'plugState': 0, 'lowTemperature': 0}}
2024-12-07 10:08:06,733:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'autoHeat': 0, 'heatState': 0, 'invOutputPower': 0}}
2024-12-07 10:08:07,013:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'ambientLightNess': 20, 'ambientLightColor': 65280, 'ambientLightMode': 0}}
2024-12-07 10:08:07,173:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'ambientSwitch': 0, 'OldMode': 0, 'pass': 0}}
2024-12-07 10:08:07,373:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'circuitCheckMode': 0, 'chargingTime': 0, 'chargingType': 0}}
2024-12-07 10:08:07,653:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'dspversion': 8453, 'hyperTmp': 2831, 'strength': 3}}
2024-12-07 10:08:07,833:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'socStatus': 0, 'gridReverse': 2, 'priceType': 0}}
2024-12-07 10:08:07,953:INFO: {'deviceId': '<deviceId>', 'method': 'BLESPP'}
2024-12-07 10:08:08,033:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'phaseCheck': 0, 'reverseState': 0, 'outputLimit': 121}}
2024-12-07 10:08:08,293:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'inputLimit': 0, 'acMode': 1, 'gridStandard': 0}}
2024-12-07 10:08:08,473:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'inverseMaxPower': 1200, 'chargingMode': 2, 'energyPower': 0}}
2024-12-07 10:08:08,654:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'lampSwitch': 1, 'BatVolt': 0, 'clusterSw': 0}}
2024-12-07 10:08:08,914:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'autoModel': 0, 'gridOffMode': 0, 'localState': 0}}
2024-12-07 10:08:09,133:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'acconnect': 1}, 'packData': [{'power': 0, 'socLevel': 48, 'sn': '<serialN>'}]}
2024-12-07 10:08:09,313:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {}, 'packData': [{'state': 0, 'maxTemp': 2801, 'totalVol': 4910, 'sn': '<serialN>'}]}
2024-12-07 10:08:09,521:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {}, 'packData': [{'maxVol': 327, 'minVol': 327, 'softVersion': 4113, 'sn': '<serialN>'}]}
2024-12-07 10:08:09,753:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {}, 'packData': [{'power': 0, 'socLevel': 50, 'state': 0, 'sn': '<serialN>'}]}
2024-12-07 10:08:09,954:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {}, 'packData': [{'maxTemp': 2801, 'totalVol': 4920, 'maxVol': 328, 'sn': '<serialN>'}]}
2024-12-07 10:08:10,193:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {}, 'packData': [{'minVol': 327, 'softVersion': 4113, 'sn': '<serialN>'}]}
[... cut ...]
2024-12-07 10:08:10,494:INFO: {'method': 'getInfo-rsp', 'deviceId': '<deviceId>', 'deviceSn': '<serialN>', 'firmwares': [{'type': 'PLC_CCO', 'version': 12556}, {'type': 'PLC_STA', 'version': 12556}]}
2024-12-07 10:08:10,495:INFO: The SF device ID is: <deviceId>
2024-12-07 10:08:10,495:INFO: The SF device SN is: <serialN>
2024-12-07 10:08:49,100:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'electricLevel': 49}}
2024-12-07 10:09:49,919:INFO: {'method': 'report', 'deviceId': '<deviceId>', 'properties': {'electricLevel': 49}}
```

cmuster, 2024-12-08:
Looks like I've made some progress: apparently  the **productKey** of my Hyper2k is not the previously known `ja72U0ha` but `gDa3tb` for some reason. Not sure if this is a (new?) revision of the Hyper or whatever. In some of the Mqtt reports its also labeled "SolarFlowPro" instead of "Hyper", but when I query the device list from the Zendure Cloud API it's still reported as  `'productName': 'Hyper 2000',`

Anyway it looks like I get at least all the telemetry now locally which is great. :) Still have to figure out controlling of the AC charging..

reinhard-brandstaedter, 2024-12-08:
Hi @cmuster thanks for confirming, with a new product ID it shouldn't be too difficult to also add support. Likely all the other behavior stays the same.
Regarding the AC Charging: there is a (recently discovered) property ```acMode``` of (all) hubs that seems to control if AC is fed in or taken (yet to be confirmed). If this property is set to 2 means output (from hub), and 1 seems to be input (to the hub). 0 is likely completely off.
You can try the variants of that setting and see what happens, if I'm correct if you set it to 1 your Hyper should start charging?

Raudi1, 2024-12-08:
> If this property is set to 2 means output (from hub), and 1 seems to be input (to the hub). 0 is likely completely off. You can try the variants of that setting and see what happens, if I'm correct if you set it to 1 your Hyper should start charging?

That's how it works for the [ioBroker adapter](https://github.com/nograx/ioBroker.zendure-solarflow). Together with setInputLimit, setOutputLimit and passMode it offers full control.

cmuster, 2024-12-08:
@reinhard-brandstaedter thank you for doing all the heavy lifting and providing solarflow control in the first place 👍
Having the possibility to control the battery entirely locally was what made me decide to buy the Zendure products (therefore **they** should really thank you for all the reverse engineering you did 😄)

It seems when I send all the properties in one message (as @Raudi1 also just confirmed) setting the ac charging works now fine. I just had to setup a seperate HomeAssistant automation for it for now, as the HA addin (https://github.com/dfigus/addon-solarflow-control), is not supporting these yet as of now.

Nice  coincidence: We're finally getting some sun today, so I can test all of this properly 😄

reinhard-brandstaedter, 2024-12-08:
> That's how it works for the [ioBroker adapter](https://github.com/nograx/ioBroker.zendure-solarflow). Together with setInputLimit, setOutputLimit and passMode it offers full control.

Now we finally know what the ominous ```inputLimit``` state report telemetry is used for? This has been around like forever with no function ;-)

Armageddit, 2024-12-27:
2024-12-27 01:39:16,260:INFO: HUB: S:-1.0W [ ], B: -1% (-1) low: 5 high: 100, V:-1.0V (-1.0), C: 0W, P:False (manual, possible), F:-1.0h, E:-1.0h, CT:OFF (333hrs) idle, H: -1W, L: -1W
2024-12-27 01:39:16,260:INFO: INV: AC:0.0W, DC:0.0W (), L: 0W (0.0W/channel) [ -1W]
2024-12-27 01:39:16,261:INFO: SMT: T:Smartmeter P:38.0W [ 47.9,47.9,48.8,49.5,50.9,40.4 ]
2024-12-27 01:39:16,261:INFO: SMT triggers limit function: 50.9 -> 40.4: executed
2024-12-27 01:39:17,251:INFO: SMT triggers limit function: 50.9 -> 39.9: skipped
2024-12-27 01:39:18,257:INFO: SMT triggers limit function: 51.0 -> 39.7: skipped
2024-12-27 01:39:19,258:INFO: SMT triggers limit function: 51.1 -> 39.9: skipped
2024-12-27 01:39:20,252:INFO: SMT triggers limit function: 51.2 -> 40.0: skipped
2024-12-27 01:39:21,256:INFO: SMT triggers limit function: 51.1 -> 40.1: skipped

i have a hyper 2000 and a SmartMeter that published multiple times per second to MQTT
this was detectet by SF--control and republished to solarflow-hub/smartmeter/homeUsage and solarflow-hub/smartmeter/homeUsageSmoothened

But the poweroutput of the Hyper 2000 is always the maximum power that I have set in the app.

There is no change or adjustment due to the smart meter values.

what is my error?

reinhard-brandstaedter, 2024-12-27:
@Armageddit hard to say with so littel information from your side.
A bit mor context would be useful, your setup? your config.ini a complete log file?

Armageddit, 2024-12-28:
think the problem is that I installed Python on Windows 11 to run the script.

but other solutions didn't work:

in WSL it doesnt run becauce dont know what xyz mean

```
*Linux_USER*@DESKTOP-*USER*:/mnt/c/Users/*USER*/Downloads/solarflow-bt-manager-master/src$ python3 solarflow-bt-manager.py -i
Traceback (most recent call last):
  File "/mnt/c/Users/*USER*/Downloads/solarflow-bt-manager-master/src/solarflow-bt-manager.py", line 4, in <module>
    from bleak import BleakClient, BleakScanner
ModuleNotFoundError: No module named 'bleak'
*Linux_USER*@DESKTOP-*USER*:/mnt/c/Users/*USER*/Downloads/solarflow-bt-manager-master/src$ pip install bleak
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
*Linux_USER*@DESKTOP-*USER*:/mnt/c/Users/*USER*/Downloads/solarflow-bt-manager-master/src$ pip3 install bleak
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

try with miniconda:

```
(base) *Linux_USER*@DESKTOP-*USER*:/mnt/c/Users/*USER*/Downloads/solarflow-bt-manager-master/src$ pip install -r requirements.txt
Requirement already satisfied: bleak in /home/*Linux_USER*/miniconda3/lib/python3.12/site-packages (from -r requirements.txt (line 1)) (0.22.3)
Requirement already satisfied: paho-mqtt>=2.0.0 in /home/*Linux_USER*/miniconda3/lib/python3.12/site-packages (from -r requirements.txt (line 2)) (2.1.0)
Requirement already satisfied: dbus-fast<3,>=1.83.0 in /home/*Linux_USER*/miniconda3/lib/python3.12/site-packages (from bleak->-r requirements.txt (line 1)) (2.24.4)
(base) *Linux_USER*@DESKTOP-*USER*:/mnt/c/Users/*USER*/Downloads/solarflow-bt-manager-master/src$ python3 solarflow-bt-manager.py -i
2024-12-28 14:33:47,715:INFO: scan for: zene
Traceback (most recent call last):
  File "/mnt/c/Users/*USER*/Downloads/solarflow-bt-manager-master/src/solarflow-bt-manager.py", line 256, in <module>
    main(sys.argv[1:])
  File "/mnt/c/Users/*USER*/Downloads/solarflow-bt-manager-master/src/solarflow-bt-manager.py", line 253, in main
    asyncio.run(run(broker=mqtt_broker, port=mqtt_port, info_only=info_only, connect=connect, disconnect=disconnect, ssid=ssid))
[... 7 lines cut ...]
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/mnt/c/Users/*USER*/Downloads/solarflow-bt-manager-master/src/solarflow-bt-manager.py", line 147, in run
    device = await BleakScanner.find_device_by_filter(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/*Linux_USER*/miniconda3/lib/python3.12/site-packages/bleak/__init__.py", line 444, in find_device_by_filter
    async with cls(**kwargs) as scanner:
               ^^^^^^^^^^^^^
  File "/home/*Linux_USER*/miniconda3/lib/python3.12/site-packages/bleak/__init__.py", line 158, in __aenter__
    await self._backend.start()
  File "/home/*Linux_USER*/miniconda3/lib/python3.12/site-packages/bleak/backends/bluezdbus/scanner.py", line 168, in start
    manager = await get_global_bluez_manager()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/*Linux_USER*/miniconda3/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 1058, in get_global_bluez_manager
    await instance.async_init()
  File "/home/*Linux_USER*/miniconda3/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 286, in async_init
    assert_reply(reply)
  File "/home/*Linux_USER*/miniconda3/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.ServiceUnknown] The name org.bluez was not provided by any .service files
(base) *Linux_USER*@DESKTOP-*USER*:/mnt/c/Users/*USER*/Downloads/solarflow-bt-manager-master/src$

```

how can i run this script with windows or homeassistantOS?

Armageddit, 2024-12-28:
[image]

Is it good?
However, the data can still be changed in the app. So the connection to the cloud is not disconnected, and no "Automatically control the charging/discharging and the power output based on the acutal consumption"

Armageddit, 2024-12-29:
ok a ubuntu stick to boot war the fix to run [solarflow-bt-manager](https://github.com/reinhard-brandstaedter/solarflow-bt-manager)
Cloud is disconnect

now runs the HomassistantOS with the addon [addon-solarflow-control](https://github.com/dfigus/addon-solarflow-control)

now the local connection works but no "Automatically control the charging/discharging and the power output based on the acutal consumption"

how can i manage this?
should i make a homeassistant automation or a nodered to control the charging and output power?

[image]

reinhard-brandstaedter, 2024-12-30:
Do you have logs of the HA addon?
I’m not maintaining the addon but sf-control of course requires the correct configuration to read the actual power usage from your smartmeter and also needs to be able to control your inverter. Without that it can’t do anything really.
So if that is given please post your ligs from sf-control startup.

Armageddit, 2025-01-02:
sollte ich über das Handy ein update per Bluetooth machen?
da gibt es bereits eine neue Firmware.

```
2025-01-02 13:31:00,239:INFO: SMT: T:Smartmeter P:1056.0W [ 1052.3,1052.3,1050.6,1050.7,1050.9,1050.6 ]
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service solarflow-control: stopping
s6-rc: info: service solarflow-control successfully stopped
s6-rc: info: service init-solarflow-control: stopping
s6-rc: info: service init-solarflow-control successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------
 Add-on: Solarflow Control
 Solarflow Control
-----------------------------------------------------------
 Add-on version: 0.10.3
 You are running the latest version of this add-on.
 System: Home Assistant OS 14.1  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.12.5
 Home Assistant Supervisor: 2024.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-solarflow-control: starting
[13:31:51] INFO: Updating Solarflow Control configuration...
s6-rc: info: service init-solarflow-control successfully started
s6-rc: info: service solarflow-control: starting
s6-rc: info: service solarflow-control successfully started
s6-rc: info: service legacy-services: starting
[13:31:52] INFO: Running Solarflow Control
s6-rc: info: service legacy-services successfully started
2025-01-02 13:31:55,609:INFO: MQTT Host: "IP-Zensiert":1883
2025-01-02 13:31:55,609:INFO: MQTT User: "ID-Zensiert"
2025-01-02 13:31:55,610:INFO: Solarflow Hub: gDa3tb/"deviceId-Zensiert"
2025-01-02 13:31:55,701:INFO: IP Address: 88.152.146.94
2025-01-02 13:31:55,701:INFO: Location: Lippetal, "zensiert", Germany
2025-01-02 13:31:55,702:INFO: Coordinates: (Lat: "zensiert", Lng: "zensiert"
2025-01-02 13:31:55,704:INFO: No config setting found for option "disable_full_discharge" in section solarflow!
2025-01-02 13:31:55,708:INFO: SF Control subscribing: solarflow-hub/"deviceId-Zensiert"/control/#
2025-01-02 13:31:55,709:INFO: Reading retained config settings from MQTT...
2025-01-02 13:31:55,709:INFO: Note: Solarflow Control persists initial configuration settings in your MQTT broker and will use those first (if found) to allow on-the-fly updates!
2025-01-02 13:31:55,709:INFO: If you want to override these values from your config.ini you need to clear those retained topics in your broker first!
2025-01-02 13:31:56,193:INFO: Connected to MQTT Broker!
2025-01-02 13:31:56,201:INFO: Found control/dischargeDuringDaytime, set DISCHARGE_DURING_DAYTIME to True
2025-01-02 13:31:56,202:INFO: Found control/sunriseOffset, set SUNRISE_OFFSET to 60 minutes
2025-01-02 13:31:56,202:INFO: Found control/sunsetOffset, set SUNSET_OFFSET to 60 minutes
2025-01-02 13:31:56,203:INFO: Found control/minChargePower, set MIN_CHARGE_POWER to 50W
2025-01-02 13:31:56,203:INFO: Found control/maxDiscahrgePiwer, set MAX_DISCHARGE_POWER to 400W
2025-01-02 13:31:56,204:INFO: Found control/batteryTargetSoCMin, set BATTERY_LOW to 5%
[... 60 lines cut ...]
2025-01-02 13:32:11,104:INFO: Rapid drop in demand detected, clearing buffer!
2025-01-02 13:32:11,106:INFO: HUB: S:0.0W [ 0.0 ], B: 31% (31) low: 5 high: 100, V:50.0V (50.0), C: 163W, P:False (manual, possible), F:-1.0h, E:-1.0h, CT:OFF (336hrs) idle, H: 0W, L: -1W
2025-01-02 13:32:11,107:INFO: INV: AC:0.0W, DC:0.0W (), L: 0W (0.0W/channel) [ -1W]
2025-01-02 13:32:11,107:INFO: SMT: T:Smartmeter P:336.0W [ 336.0,336.0 ]
2025-01-02 13:32:11,107:INFO: SMT triggers limit function: 336.0 -> 336.0: executed
2025-01-02 13:33:05,719:INFO: Triggering telemetry update: iot/gDa3tb/"deviceId-Zensiert"/properties/read
2025-01-02 13:34:05,720:INFO: Triggering telemetry update: iot/gDa3tb/"deviceId-Zensiert"/properties/read
2025-01-02 13:34:05,914:INFO: HUB: S:0.0W [ 0.0 ], B: 31% (31|32) low: 80 high: 100, V:49.4V (49.4|49.4), C: 0W, P:False (manual, possible), F:-1.0h, E:-1.0h, CT:OFF (336hrs) idle, H: 0W, L:339W
2025-01-02 13:34:05,915:INFO: INV: AC:0.0W, DC:0.0W (), L: 0W (0.0W/channel) [ -1W]
2025-01-02 13:34:05,915:INFO: SMT: T:Smartmeter P:339.0W [ 329.4,329.4,330.6,332.9,336.2,340.2 ]
[... cut ...]
2025-01-02 13:36:00,343:INFO: SMT triggers limit function: 320.2 -> 309.3: skipped
2025-01-02 13:36:01,343:INFO: SMT triggers limit function: 318.5 -> 307.0: skipped
2025-01-02 13:36:02,344:INFO: SMT triggers limit function: 316.8 -> 304.7: skipped
2025-01-02 13:36:03,337:INFO: SMT triggers limit function: 315.1 -> 302.5: skipped
2025-01-02 13:36:04,344:INFO: SMT triggers limit function: 313.6 -> 300.3: skipped
2025-01-02 13:36:05,341:INFO: SMT triggers limit function: 312.0 -> 298.1: skipped
```

dfigus, 2025-01-09:
I guess he is using the Hyper which includes an inverter - hence no OpenDTU/AhoyDTU. @reinhard-brandstaedter Is solarflow-control able to control the integrated inverter as well instead of a DTU?

ThorstenNauroth, 2025-01-11:
Hi, i can't geht the informations from my Hyper2000. The solarflow-bt-manager.py says "scan for: zen**p**" but it ist "zen**e**". How can i change this?
[image]

cmuster, 2025-01-11:
@ThorstenNauroth you have to adjust your productId accordingly. See solarflow-bt-manager.py L134-143

unExpectedBracket, 2025-01-25:
> I guess he is using the Hyper which includes an inverter - hence no OpenDTU/AhoyDTU. [@reinhard-brandstaedter](https://github.com/reinhard-brandstaedter) Is solarflow-control able to control the integrated inverter as well instead of a DTU?

Yes, the Hyper2000 has the inverter included. I also came across this issue. I put the Hyper's  report 

`/gDa3tb/x3J111pe/properties/report`

into the DTU mqtt Channel in the config and the log from the script shows 0.0 from the Inverter. 

` INV: AC:0.0W, DC:0.0W (), L: 0W (0.0W/channel) [ -1W]`

Since it is dark I thought that was okay. But now I remember it didn't ask for an accessor like in the tasmota section. So the script will not control anything because it thinks there is no inverter?

didlawowo, 2025-02-19:
Hello, i'm trying to do the same thing on my side.
i'm able to connect into mqtt on zendure cloud infra. now i'm looking for information to setup power output / battery level

zaubara, 2025-02-21:
Hi all,
yes, the Hyper 2000 has all that included. After setting everything up, it's possible to control it with a publish to properties/write, acMode 1 to charge, acMode 2 to discharge, with limits, and setting the outputLimit to 0 to idle.
So there a few few properties missing for the Hyper to control it fully within HA, and there's no DTU, so this would need to be disabled -- I managed to get the smart meter included, but with no DTU configured, theres no automatic charging/discharging -- this would need to be updated as well, I think.

didlawowo, 2025-02-22:
when you say smart meter you're speaking about satellite monitor CT ?  do you know how to retrieve information from him ?

cbergmann, 2025-04-06:
Hi, after reading this thread and looking into the code I wantend to consolidate the things that I found and get feedback on the next steps to support Hyper 2000:

1. The `${productKey}` of Some Hyper 2000 seem to be inconsistent. Some use `ja72U0ha` and some use `gDa3tb`. Mine seems to be of the last variant.
2. When configuring solarflow-controll correctly all values of the `properties` attribute of the JSON object posted by the Hub to `/${productKey}/${deviceKey}/properties/report` will be made available as dedicated values in topics of the form `solarflow-hub/${deviceKey}/telemetry/${property}`.
3. Also the output of the Hyper 2000 to the grid can be configured by setting `solarflow-hub/${deviceKey}/telemetry/outputLimit`.
4. In this state the [limitHomeInput](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/src/solarflow/solarflow-control.py#L336C5-L336C19) function will be triggered on each update of hub or smart meter.
5. When no inverter is configured however solarflow-controll will not adopt the output power because of [this check](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/src/solarflow/solarflow-control.py#L347) in the code.

If I understand this correctly the next steps could be to implement a "Hyper2000"  inverter in [dtus.py](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/src/solarflow/dtus.py). That implementation would get the relevant values from `solarflow-hub/${deviceKey}/telemetry` and set `solarflow-hub/${deviceKey}/telemetry/outputLimit` whenever `setLimit` is called.

Before I start implementing this Inverter could please someone with better insight into the code confirm that this is a valid path to follow.

Kind regards
Clemens

cbergmann, 2025-04-06:
fyi here are all of the `${property}`s with some example values that I have seen:

```JSON
{
    "acMode":2,
    "ambientLightColor":65280,
    "ambientLightMode":0,
    "ambientLightNess":20,
    "ambientSwitch":0,
    "autoHeat":0,
    "autoModel":0,
    "BatVolt":0,
    "blueOta":1,
    "buzzerSwitch":0,
    "chargingMode":2,
    "chargingTime":0,
    "chargingType":0,
    "circuitCheckMode":0,
    "clusterSw":0,
    "ctOff":0,
    "dspversion":8459,
    "electricLevel":4,
    "energyPower":0,
    "gridInputPower":0,
    "gridOffMode":0,
    "gridReverse":2,
    "gridStandard":0,
    "heatState":0,
    "hubState":0,
    "hyperTmp":2831,
    "inputLimit":0,
    "inputMode":0,
    "inverseMaxPower":800,
    "invOutputPower":0,
    "lampSwitch":1,
    "localState":0,
    "lowTemperature":0,
    "masterhaerVersion":0,
    "masterSoftVersion":8471,
    "masterSwitch":1,
    "minSoc":50,
    "OldMode":0,
    "outputHomePower":0,
    "outputLimit":200,
    "outputPackPower":8,
    "outputPackPowerCycle":8,
    "packInputPower":0,
    "packNum":1,
    "packState":1,
    "pass":0,
    "phaseCheck":0,
    "plugState":0,
    "pvBrand":0,
    "remainInputTime":59940,
    "remainOutTime":59940,
    "reverseState":0,
    "socSet":1000,
    "socStatus":0,
    "solarInputPower":8,
    "solarPower1":8,
    "solarPower1Cycle":8,
    "solarPower2":0,
    "strength":3,
    "VoltWakeup":0,
    "wifiState":1,
    "packData":[{"power":13,"batcur":3,"socLevel":4,"state":1,"maxTemp":2911,"totalVol":4390,"maxVol":293,"minVol":292,"softVersion":4113,"sn":"REDACTED"}]
}
```

DaPeace, 2025-04-09:
@Armageddit do you use the solarflow-addon? Did i understand it right, that with hyper 2000 and this addon i cant set the discharge? im tryin around right now. I want to control the stuff via node red but discharging is not working at the moment. The addon is running and i use the telemetry-values to monitor the battery but when i set acmode to 2 and set the outputlimit it does not work. the battery and the inverter are new and not used with zendure cloud. Ive only discharged the battery with the app and charged via node red today but i cant get discharging to work.

linuxstony, 2025-04-11:
@dfigus do I see this correctly that you can control the hyper on the DC side only with that integration and not on the AC side or should I have full control over the Hyper with your HA-Integration?

cbergmann, 2025-04-11:
@linuxstony thanks for bringing that up. While thinking about how to implement my suggestion I also stumbled upon that. Currently I can control the output of the Hyper to the Home Network by changing `outputLimit`. When the battery is not full the internal firmware seems to automatically redirect additional power to the batteries.
I am not certain if I can control that via any of the MQTT topics. If that is possible we can implement the DTU und Hub side of Hyper 2000 independently. If not we have to think about how these two implementations should work together. Currently the Hub implementation already [controlls](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/src/solarflow/solarflow.py) `outputLimit` so in this case a "dummy" implementation of DTU for the Hyper2000 might be one option. This one could only override [ready()](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/src/solarflow/dtus.py#L59) with only returning `true` so the Hub implemenation can take over.

dfigus, 2025-04-11:
> [@dfigus](https://github.com/dfigus) do I see this correctly that you can control the hyper on the DC side only with that integration and not on the AC side or should I have full control over the Hyper with your HA-Integration?

I don't own a hyper, I'm running it with a Hub2k and a Hoymiles inverter via OpenDTU.

didlawowo, 2025-04-12:
> > [@Armageddit](https://github.com/Armageddit) do you use the solarflow-addon? Did i understand it right, that with hyper 2000 and this addon i cant set the discharge? im tryin around right now. I want to control the stuff via node red but discharging is not working at the moment. The addon is running and i use the telemetry-values to monitor the battery but when i set acmode to 2 and set the outputlimit it does not work. the battery and the inverter are new and not used with zendure cloud. Ive only discharged the battery with the app and charged via node red today but i cant get discharging to work.
> 
> at the moment i use `iot/gDa3tb/XXX/properties/write`
> 
> Charge:
> 
> ```
> {
>   "properties": {
>     "acMode": 1
>   }
> }
> ```
> 
> ```
> {
>   "properties": {
>     "inputLimit": 188
>   }
> }
> ```
> 
> Discharge:
> 
> ```
> {
>   "properties": {
>     "acMode": 2
>   }
> }
> ```
> 
> ```
> {
>   "properties": {
>     "outputLimit": 188
>   }
> }
> ```
> 
> [image]
> 
> my main function to control the hyper 2000:
> 
> ```
> let zahler = msg.payload;
> let powerset;
> let acmode
> let npow =-150  //Einspeisen
> let pow = -30   //Laden
> 
> let battery_pz = flow.get('battery_%') || 0;
> let battery_input = flow.get('battery_input') || 0;
> let battery_output = flow.get('battery_output') || 0;
> let powerset_alt = flow.get('powerset') || 0;
> 
> powerset = battery_input - battery_output - zahler +7 ;
> 
> if (powerset >= 1200){powerset = 1200}
> if (powerset <= -600){powerset = -600}
> let differenz = Math.abs(powerset_alt-powerset);
> 
> if (battery_pz >= 70) { npow = -60 } // Voller akku früher Einspeißen
> 
> 
> if (differenz >= 3) {
>     flow.set("powerset", powerset);
>     if (powerset <= npow) {             //entladen
>         acmode = 2
>         let powersetn = (powerset *-1);
>         msg.payload = { "properties": { "outputLimit": powersetn } };
>     }
>     else if (powerset >= pow) {          // laden
>         acmode = 1
>         msg.payload = { "properties": { "inputLimit": powerset } };
>     }
>     else {                              // standbye
>         acmode = 2;
>         powerset = 0;
>         msg.payload = { "properties": { "outputLimit": powerset } };
>     }
> 
>     flow.set("acmode_set", acmode);
>     var msg2 = { payload: acmode };
>     msg2.payload = { "properties": { "acMode": acmode } };
>     node.status({ text: zahler + " set:" + powerset + " / " + differenz + " acmode:" + acmode + " Bat:" + battery_pz +"%"});
>     
>     return[msg, msg2];
> }
> node.status({ text: zahler + " set:" + powerset_alt + " / " + differenz + " Skip"+ " Bat:" + battery_pz +"%"});
> ```
> 
> there are some buttons to manual test debug and watch if all works [image]
> 
> [image]
> 
> I hope this is to understand [@DaPeace](https://github.com/DaPeace) and all people with the same problem
> 
> PS: I only put my Hyper 2000 in the wall, not in the solar panel

hello,
i have build my own solution to pilot my zendure battery, i 'm not able to change the battery power charge.

could you share how did you connect to the hyper2000 ?  
what is flow ? 

this is my logs:
 Applying charging decision: HyperChargingDecision(should_charge=True, power=1200, reason='Charge solaire été')
ic| topic: 'iot/gDa3tb/EHe6x1wB/function/invoke'
    payload: ('{"arguments": [{"autoModelProgram": 2, "autoModelValue": {"chargingType": 1, '
              '"chargingPower": 1200, "freq": 0, "outPower": 0}, "msgType": 1, "autoModel": '
              '0}], "deviceKey": "EHe6x1wB", "function": "deviceAutomation", "messageId": '
              '1, "timestamp": 1744450819}')
2025-04-12 11:40:19.899 | INFO     | providers.zendureMqtt:_ensure_connected:181 - ✅ Connexion MQTT active
ic| self.client_context: <aiomqtt.client.Client object at 0x1088e7590>
ic| client: <aiomqtt.client.Client object at 0x1088e7590>
ic| result: None
2025-04-12 11:40:20.039 | INFO     | providers.zendureMqtt:publish:118 - 📤 Message publié sur iot/gDa3tb/EHe6x1wB/function/invoke avec résultat: None
2025-04-12 11:40:20.039 | INFO     | controllers.hyperMqtt:control_battery_power:182 - 🔋 Charge configurée à 1200W
ic| payload: '{"properties": {"inputLimit": 1200}}'
2025-04-12 11:40:20.041 | INFO     | providers.zendureMqtt:_ensure_connected:181 - ✅ Connexion MQTT active
ic| self.client_context: <aiomqtt.client.Client object at 0x1088e7590>
ic| client: <aiomqtt.client.Client object at 0x1088e7590>
ic| result: None
2025-04-12 11:40:20.167 | INFO     | providers.zendureMqtt:publish:118 - 📤 Message publié sur iot/gDa3tb/EHe6x1wB/properties/write avec résultat: None
2025-04-12 11:40:20.168 | INFO     | controllers.hyperMqtt:set_input_limit:64 - ⚡ Limite d'entrée définie à 1200W
2025-04-12 11:40:20.168 | INFO     | providers.zendureMqtt:_ensure_connected:181 - ✅ Connexion MQTT active
ic| self.client_context: <aiomqtt.client.Client object at 0x1088e7590>
ic| client: <aiomqtt.client.Client object at 0x1088e7590>
ic| result: None
2025-04-12 11:40:20.294 | INFO     | providers.zendureMqtt:publish:118 - 📤 Message publié sur iot/gDa3tb/EHe6x1wB/properties/write avec résultat: None
2025-04-12 11:40:20.294 | INFO     | controllers.hyperMqtt:set_output_limit:52 - ⚡ Limite de sortie définie à 0W
2025-04-12 11:40:20.294 | INFO     | providers.zendureMqtt:_ensure_connected:181 - ✅ Connexion MQTT active
ic| self.client_context: <aiomqtt.client.Client object at 0x1088e7590>
ic| client: <aiomqtt.client.Client object at 0x1088e7590>
ic| result: None

i suspect problem, cause i didn't see any result when i'm publish on topic . but suscribe / listen is ok

DaPeace, 2025-04-13:
Thanks guys. For 2 days now i can control the hyper 2000 and charge it with the power of my other pv-system via ac-mode. Another question, i am controlling it "by hand" via the mbus-smart meter my provider installed at the house. Did someone find out how the shelly is talking to the hyper2000 and controlling it? I wonder if it wouldnt be easier to emulate a shelly em3 and write the data to the hyper and let him do his thing?

Raudi1, 2025-04-13:
The HA integration is emulating the Shelly local communication.
https://github.com/fireson/zendure-ha

Armageddit, 2025-04-13:
@didlawowo
this is for Node-RED
I wish I had a better solution!
But my solution works offline without a cloud

Armageddit, 2025-04-13:
@DaPeace 
I'm not entirely sure but to me it sounds like the Zendure cloud is communicating with the Shelly cloud. so the ESP with infrared communication on the electricity meter is not compatible as it uses the local MQTT server.

DaPeace, 2025-04-13:
> The HA integration is emulating the Shelly local communication. https://github.com/fireson/zendure-ha

Looks like the integration is not working with AC-Charge.
@Armageddit ok. Ill let in run until my solarcells arrive hopefully this week. Thank god i did not fully bet on the ac-charging mode. In the app you have to select this mode. Im not sure if that is working with the shelly too.

der-picknicker, 2025-05-09:
Hello everyone,

I also have a Hyper 2000 (including two AB2000S batteries)...
I also don't have a DTU running (yet).
Theoretically, the whole thing should work without a DTU/additional inverter with the zero-feedback option.
Unfortunately, I discovered that
1. the zero-feedback option isn't working
2. the battery management isn't working as intended
3. This probably has nothing to do with this, but I have different settings/values ​​in the Solarflow Control configuration and in the MQTT data from solarflow control!
4. I actually configured "NO Discharge during Daytime," which unfortunately doesn't work either.

Since I've read up a bit on this topic, I simply lowered the output limit to 10W this morning. The hub actually charged the excess energy into the battery up to the maximum limit.
When the battery was full, however, the Hyper throttled down and stopped feeding energy into the home grid, even though it needed it.
At some point this afternoon, I set the output limit back to 1200W, and then the Hyper drained the battery completely again (at 1200W), even though it didn't actually need that much power!

Unfortunately, I have no programming expertise and can only provide support with my configuration and log files. I'm also happy to test various things.
Please let me know what data I should provide or what I should try, and I'll try it.

PS: In addition to the Hyper2000 (PV modules facing west), I also have a Hoymiles Inverter running (modules facing east) – I want to integrate it into HomeAssistant with an OpenDTU – unfortunately, I haven't gotten around to it yet – but I thought it should actually be possible to get the whole thing running without an additional inverter or DTU.

As I wrote before: please let me know if I can help in any way, be it with data (the configuration or the logs) or to test anything with the Hyper2000...

Sunny greetings vrom Germany
Philipp alias the-picknicker :o)

martin-nrw-85, 2025-05-09:
> Hello everyone,
>
> I also have a Hyper 2000 (including two AB2000S batteries)...
> I also don't have a DTU running (yet).
> Theoretically, the whole thing should work without a DTU/additional inverter [...]

Hi magst du mir einmal auf deutsch deine Problematik und deinen Wünsch wo du hin möchtest schreiben. 
Ich habe meinen Hub1200 lokal an einen mqtt Broker laufen ohne diese Zendure Cloud. Daher habe ich lediglich meine latenzzeiten im Heim-WLAN. Damit komm ich sehr nahe an eine Null-Watt Einspeisung ran.

der-picknicker, 2025-05-10:
> Hi magst du mir einmal auf deutsch deine Problematik und deinen Wünsch wo du hin möchtest schreiben. Ich habe meinen Hub1200 lokal an einen mqtt Broker laufen ohne diese Zendure Cloud. Daher habe [...]

Hallo Martin,

gerne schreib ich auf deutch (einfacher für mich)...
Also ich bin schon von Anfang an ohne Cloud unterwegs.
Ich habe den Hyper per solarflow-control in mein Homeassistant integriert (mit dem Addon von dfigus).
Einen Shelly EM3, der den Gesamtnetzverbrauch misst, habe ich auch integriert.
Somit könnte meiner Meinung nach zumindest theoretisch schon eine Nulleinspeisung realisiert werden, da das solarflow-control ja jetzt weiß, wieviel Energie vom Hyper ins Heimnetz muss und der Rest kann in den Akku.
Ich habe zwar auch noch zwei zusätzliche WR, die erstmal nix mit dem Hyper zu tun haben, aber nach meinem Verständnis müsste das erstmal garnicht so relevant sein für die Regelung, da der EM3 ja das Bezugsnetz misst und somit mitbekommen würde, wenn die zusätzlichen WR Energie ins Hausnetz einspeisen. dann würde der Hyper ggf. die "überschüssige" Energie sowieso in den Akku laden.
Falls die Info, was von den zusätzlichen WR kommt, notwendig ist, da hängt auch noch ein Shelly dran, so dass ich im Moment zummindest mal sehen kann, was da kommt.
DTU wäre auch noch denkbar, da es sich um Hoymiles WR handelt - da hänge ich aber im Moment noch fest...

Wenn ich solarflow-control noch erweitern/ergänzen muss oder was "drum herum" bauen muss, damit das oben Beschriebene funktioniert, brauche ich dafür Hilfe - ich bin was Homeassistant, MQTT, etc. angehet, eher Anwender.
Aber in die Cloud möchte ich die Geräte eigentlich auch nicht bringen!

Gruß
Philipp
alias der-picknicker

PS: Da Bilder mehr als 1000 Worte sagen, hier noch zwei Screenshots, die relativ gut zeigen, wie mein Setup aussieht und was passiert (ader eben nicht passiert)

[image]
[image]

der-picknicker, 2025-05-10:
Ok, ich hatte gehofft drum herum zu kommen selbst was programmieren zu müssen...
Aber dann muss ich mich wohl doch Mal damit beschäftigen!
Eine Frage hab ich tatsächlich: Wofür benötigtst Du den mqtt-Topic-mapper bzw. was macht der?
Warum nicht in Homeassistent alles laufen haben?
Und es wäre cool, wenn Du Dein Script zur Verfügung stellen könntest, damit ich irgendwie Mal nen groben Schimmer hab, wie ich mit der Geschichte loslegen kann!?!

der-picknicker, 2025-05-12:
Moin zusammen,
habe noch ein Update/Nachtrag:
Ich habe nochmal die komplette Diskussion hier durchgelesen und dachte mir, bevor ich mich mehrere Tage mit der Programmierung eines eigenen Scripts beschäftige, geb ich dem DTU nochmal ne Chance...
Gestern Abend habe ich dann meinen openDTU zum Laufen bekommen (nach diversen Kabel- und Lötproblemen) und soeben diesen in Homeassistant bzw. solarflow-control integriert.
Auf den ersten Blick sieht das schonmal ganz gut aus - hier ein "Beispiel" Log aus solarflow-control:

> 2025-05-12 07:30:10,184:INFO: HUB: S:86.7W [ 86.6,86.6,86.7,87.0,86.8,86.7 ], B: 12% (12|11) low: 10 high: 98, V:47.2V (47.2|47.2), C: 88W, P:False (manual, possible), F:228.5h, E:-1.0h, CT:OFF (168hrs) idle, H: 0W, L: 0W
2025-05-12 07:30:10,184:INFO: INV: AC:1020.8W, DC:1074.3W (265.4|259.3|277.2|271.3), L:1040W (260.0W/channel) [1600W]
2025-05-12 07:30:10,184:INFO: SMT: T:ShellyEM3 P:-361.3W [ -340.6,-340.6,-335.2,-356.9,-369.6 ]
2025-05-12 07:30:10,184:INFO: Direct connected panels (504.1W) can't cover demand (650.1W), trying to get 146.0W from hub.
2025-05-12 07:30:10,184:INFO: The current max direct channel power 257.8W is close to the current channel limit 260.0W, trying to get more from direct panels.
2025-05-12 07:30:10,184:INFO: Based on time, solarpower (86.7W) minimum charge power (125W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2025-05-12 07:30:10,185:INFO: Setting inverter output limit to 1032W (1 min moving average of 258W x 4)
2025-05-12 07:30:10,185:INFO: Grid feed in from unknown: 369.6!
2025-05-12 07:30:10,185:INFO: Sun: 05:47 - 20:59 Demand: 650.1W, Panel DC: (504.1W), Hub DC: (515.6W), Inverter Limit: 1032.0W, Hub Limit: 0.0W
2025-05-12 07:31:10,118:INFO: Triggering telemetry update: iot/gDa3tb/xxx/properties/read
2025-05-12 07:32:10,118:INFO: Triggering telemetry update: iot/gDa3tb/xxx/properties/read
2025-05-12 07:32:10,184:INFO: HUB: S:89.9W [ 89.6,89.6,89.6,89.5,89.7,89.9 ], B: 12% (12|11) low: 10 high: 98, V:47.2V (47.2|47.2), C: 89W, P:False (manual, possible), F:228.5h, E:-1.0h, CT:OFF (168hrs) idle, H: 0W, L: 0W
2025-05-12 07:32:10,184:INFO: INV: AC:1025.7W, DC:1079.5W (267.3|261.1|279.0|271.8), L:1032W (258.0W/channel) [1600W]
2025-05-12 07:32:10,184:INFO: SMT: T:ShellyEM3 P:-285.4W [ -296.0,-296.0 ]
2025-05-12 07:32:10,184:INFO: Direct connected panels (506.4W) can't cover demand (729.5W), trying to get 223.1W from hub.
2025-05-12 07:32:10,185:INFO: The current max direct channel power 258.3W is close to the current channel limit 258.0W, trying to get more from direct panels.
2025-05-12 07:32:10,185:INFO: Based on time, solarpower (89.9W) minimum charge power (125W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2025-05-12 07:32:10,185:INFO: Not setting inverter output limit as it is identical to current limit!
2025-05-12 07:32:10,185:INFO: Grid feed in from unknown: 296.0!
2025-05-12 07:32:10,186:INFO: Sun: 05:47 - 20:59 Demand: 729.5W, Panel DC: (506.4W), Hub DC: (519.1W), Inverter Limit: 1036.0W, Hub Limit: 0.0W
2025-05-12 07:32:47,714:INFO: HUB: S:89.0W [ 89.7,89.7,89.3,89.4,89.5,89.0 ], B: 12% (12|11) low: 10 high: 98, V:47.2V (47.3|47.2), C: 90W, P:False (manual, possible), F:228.5h, E:-1.0h, CT:OFF (168hrs) idle, H: 0W, L: 0W
2025-05-12 07:32:47,715:INFO: INV: AC:1027.8W, DC:1081.8W (267.6|261.6|279.1|273.5), L:1032W (258.0W/channel) [1600W]
2025-05-12 07:32:47,715:INFO: SMT: T:ShellyEM3 P:-345.1W [ -316.8,-316.8,-333.8 ]
2025-05-12 07:32:47,715:INFO: Direct connected panels (508.4W) can't cover demand (694.0W), trying to get 185.6W from hub.
2025-05-12 07:32:47,715:INFO: The current max direct channel power 259.8W is close to the current channel limit 258.0W, trying to get more from direct panels.
2025-05-12 07:32:47,715:INFO: Based on time, solarpower (89.0W) minimum charge power (125W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2025-05-12 07:32:47,715:INFO: Setting inverter output limit to 1040W (1 min moving average of 260W x 4)
2025-05-12 07:32:47,715:INFO: Grid feed in from unknown: 333.8!
2025-05-12 07:32:47,716:INFO: Sun: 05:47 - 20:59 Demand: 694.0W, Panel DC: (508.4W), Hub DC: (519.4W), Inverter Limit: 1040.0W, Hub Limit: 0.0W
2025-05-12 07:32:47,716:INFO: SMT triggers limit function: -316.8 -> -333.8: executed
2025-05-12 07:32:47,717:INFO: SMT triggers limit function: -317.5 -> -332.3: skipped
2025-05-12 07:32:47,718:INFO: SMT triggers limit function: -317.9 -> -330.9: skipped

Ich verwende, nachdem sich Daniel hier schon beteiligt hat, inzwischen den Fork von ihm:

[image]
Ob das jetzt für mich / meinen Fall einen Unterschied macht, weiß ich nicht...
@dfigus Vielleicht kannst Du ja selbst dazu was sagen bzw. schreiben?

Ich werde das jetzt jedenfalls erstmal so laufen lassen und beobachten - mein Wunschtraum wäre jedenfalls, dass ich so jetzt schon quasi das Optimum habe um möglichst keine Energie zu "verschenken" bzw. nur die, die ich sowieso nicht selbst nuten kann ;-)

Sonnige Grüße und einen schönen Wochenstart
Philipp
alias der-picknicker

der-picknicker, 2025-05-12:
Guten Abend zusammen,

ein weiteres Update von mir:
Zuerst sah das heute morgen gut aus - alles was vom Hyper kam wurde in die Akkus geladen (da stand auch genug Energie vom Hoymiles zur Verfügung um den Eigenbedarf zu decken, also erstmal alles ok soweit...
Als der Akku dann voll geladen war (nach 13:00 Uhr) wurde jedoch der Hyper auf 0W runter gedrosselt, obwohl da dann der Bedarf schon größer war als der Hoymiles liefern konnte!
Hat jemand dafür eine Erklärung?

[image]

der-picknicker, 2025-05-14:
Guten Morgen zusammen,

wieder ein Update von mir...
Ich habe mir gestern Abend nochmal mit ein paar Einstellungen herumgespielt (muss man wirklich leider so sagen, da ich nicht wirklich verstanden habe, welcher "Schalter" was macht).
Hier habe ich folgendes eingestellt:

[image]

Und dann konnte ich hier manuell 250W eintragen, so dass der Hyper über Nacht konstant 250W aus dem Akuu entnommen hat:

[image]
Bevor ich die Einstellungen geändert hatte konnte ich keinen Wert eintragen bzw. das wurde immer wieder durch 0W (bzw. 1200W bis der Akku voll war) überschrieben.

Jetzt kann solarflow-control jedoch die 250W nicht mehr ändern - das scheint jetzt fest eingestellt zu sein:

> 2025-05-14 07:33:03,301:INFO: HUB: S:93.1W [ 93.3,93.3,92.9,93.1,93.1,93.1 ], B: 30% (30|31) low: 10 high: 100, V:48.5V (48.5|48.4), C:-162W, P:False (manual, possible), F:276.5h, E:-1.0h, CT:ON (0hrs) charging, H:249W, L:250W
2025-05-14 07:33:03,302:INFO: INV: AC:921.5W, DC:969.9W (237.3|231.7|256.3|244.6), L:1600W (400.0W/channel) [1600W]
2025-05-14 07:33:03,302:INFO: SMT: T:ShellyEM3 P:-728.0W [ -732.6,-732.6,-735.8,-736.0,-737.1,-737.3 ]
2025-05-14 07:33:03,302:INFO: Direct connected panels (452.5W) can cover demand (184.2W)
2025-05-14 07:33:03,302:INFO: Hub has recently adjusted limit, need to wait until it is set again! Current limit: 250, new limit: 0.0
2025-05-14 07:33:03,302:INFO: [DRYRUN] Setting inverter output limit to 960W (1 min moving average of 240W x 4)
2025-05-14 07:33:03,302:INFO: Grid feed in from unknown: 737.3!
2025-05-14 07:33:03,302:INFO: Sun: 05:44 - 21:02 Demand: 184.2W, Panel DC: (452.5W), Hub DC: (469.0W), Inverter Limit: 960.0W, Hub Limit: 250.0W
2025-05-14 07:33:16,297:INFO: SMT triggers limit function: -741.6 -> -730.5: skipped
2025-05-14 07:33:16,299:INFO: SMT triggers limit function: -741.3 -> -730.4: skipped
2025-05-14 07:33:16,301:INFO: SMT triggers limit function: -740.5 -> -729.7: skipped

Ich gehe inzwischen davon aus, dass ich einfach nur die richtigen Einstellungen vornehmen muss, damit das hier alles sauber läuft...
Kann mir bitte jemand helfen?

Sonnige Grüße
Philipp

dfigus, 2025-05-14:
@der-picknicker Ich glaube ich komme so langsam dahinter, was du hier versuchst - aber das funktioniert meiner Meinung nach nicht mit Reinhards solarflow-control.

Dein Setup scheint folgendermaßen zu sein

- Hyper mit einem Teil der Panels und direkt in die erste Steckdose geht (der Hyper hat ja ein Wechselrichter eingebaut, anders als beim Hub)
- Hoymiles Wechselrichter mit dem anderen Teil der Panels, der in eine zweite Steckdose geht

Reinhards solarflow-control kann aber nur einen Hub mit einem direkt angeschlossenem Wechselrichter steuern. In deinem Fall versucht solarflow-control den Hoymiles runterzuregeln, da er eine Einspeisung entdeckt - der steht aber schon auf 0 Watt, da die Panele dort sowieso nach Sonneruntergang nichts produzieren. Wenn die direkt an den Wechselrichter angeschlossenen Panele nichts mehr produzieren, setzt Solarflow-control das Output Limit von Hub/Hyper auf den Maximalwert, damit alles aus der Batterie kommt. Genau das Phänomen siehst du hier.

Dein Setup ist einfach nicht von Reinhard supported. Du musst deine Regelung selbst programmieren.

Damit ein Setup mit einem Hyper - OHNE zusätzlichen Hoymiles Wechselrichter - korrekt gesteuert werden kann, muss eine virtuelle DTU für den Hyper Wechselrichter implementiert werden.

der-picknicker, 2025-05-14:
> [@der-picknicker](https://github.com/der-picknicker) Ich glaube ich komme so langsam dahinter, was du hier versuchst - aber das funktioniert meiner Meinung nach nicht mit Reinhards [...]

Hallo Daniel,

vielen Dank für Deine Rückmeldung!
Ja, genau - so sieht mein Setup aus...
Und tatsächlich hatte ich inzwischen schon verstanden, dass der Hyper so quasi nicht unterstützt wird, hatte aber auch insgeheim gehofft, dass entweder Du selbst quasi die Unterstützung in Deinen Fork eingebaut hast oder darüber nachdenkst dies zu tun (hattest Du nicht auch hier weiter oben sowas in der Art geschrieben?).
Oder eben, dass mir hier jemand anderweitig helfen kann (mit einem Tipp, wie man vielleicht eine solche Unterstützung "einbaut" oder "drum herum baut" - vielleicht direkt in HA oder wie auch immer...
Dass ich wohl selbst was programmieren muss, hatte oben auch schon jemand geschrieben und ich hab mir daher mal den Code von Reinhard angeschaut - da blicke ich allerdigs leider überhaupt nicht durch! Und davon abgesehen, dass ich noch keine eine Zeile Python programmiert habe, liegen meine Kenntnisse zur textbasierten Programmierung rund 30 Jahre zurück und basieren auf ein paar Zeilen Basic und Pascal, falls das hier überhaupt noch jemandem was sagt :o)
Eventuell wäre ein gangbare Variante das Ganze mit Node-Red zu bauen, damit habe ich zwar auch noch nicht programmiert, aber mit grafischer Programmierung hatte ich immerhin ein paar Berührungspunkte in den letzten 15 Jahren - vielleicht kann ja jemand hier sagen, ob das damit machbar wäre oder eher nicht!?!
Weitere Möglichkeit scheint noch openDTU on Battery zu sein (da bin ich jetzt noch drüber gestolpert), da ist mein Setup aber offensichtlich auch nicht direkt vorgesehen :(
Ich bin daher aktuell tatsächlich etwas ratlos, was ich machen soll und hab auch schonmal ganz kurz drüber nachgedacht, die Zendure-Cloud zu nutzen (aber auch nur gaaaaaaaaanz kurz).

Liebe Grüße
Philipp

DaPeace, 2025-05-15:
Hi Philipp, ich hab das solarflow-control relativ schnell zu den Akten gelegt weil es mit Nodered eigentlich einfacher geht. Wenn du MQTT umgeleitet hast, brauchst du eigentlich nur mit "acmode" und "output/inputlimit" als json-string die Werte schreiben und der Wechselrichter macht dann. Mit der aktuellen Firmware wurden auch paar Probleme beseitigt die vorher problematisch waren z.B. Vollabschaltung bei erreichen des Ladestandes. Wichtig ist das du den Hausverbrauch/Bezug/Überschuss irgendwo misst und im NodeRed oder via MQTT verfügbar hast und dann kannst du da drauf direkt reagieren.

der-picknicker, 2025-05-15:
> Hi Philipp, ich hab das solarflow-control relativ schnell zu den Akten gelegt weil es mit Nodered eigentlich einfacher geht. Wenn du MQTT umgeleitet hast, brauchst du eigentlich nur mit "acmode" und [...]

Hallo DaPeace,
vielen Dank für Deinen Beitrag!
Gedanklich war ich eigentlich schon so weit - wenn Du jetzt auch noch schreibst, dass es sogar einfacher geht mit NodeRed, werde ich das jetzt wohl mal angehen (Addon in HA ist schon installiert)...

Da hab ich aber gleich mal noch zwei Fragen:
Was meinst Du mit "Wenn du MQTT umgeleitet hast..."?
Wenn ich das richtig verstanden habe müsste das ja schon Reinhards solarfow-bt-manager erledigt haben, oder?
Also ich hab jedenfalls die MQTT-Daten des Hyper2000 im Homeassistant - kann natürlich sein, dass die weg sind, wenn ich solarflow-control deaktiviere!?

Zweite Frage: Kannst Du mir Dein "Node" für die Regelung zukommen lassen?
Ich fange da wirklich bei "Null" an und bin über jede Hilfe, Beispiele, Lösungsansätze, etc. dankbar :)
Dass ich für meinen Fall natürlich nix Fertiges bekomme und selbst auch noch was tun muss ist mir klar, aber wenn ich schonmal eine Grundlage hätte, worauf ich dann aufbauen kann, wäre das um Einiges einfacher umzusetzen, denke ich!

Viele Grüße
Philipp

[5 more replies]