reinhard-brandstaedter/solarflow-control #273: erroneous grid feed from the battery

xsived, 2024-07-05:
Hi Reinhard,

This evening I detected a strange behavior of sf-control. It was excessively feeding the grid from the battery. The software was working, but it didn't limit the inverter to stop the leakage :-)
I did set manually the inverter limit to 0 through openDTU and after that sf-control started working properly again.
I'm attaching the log entries around the start and the end of the misbehavior.
[feed_to_grid.txt](https://github.com/user-attachments/files/16113830/feed_to_grid.txt)

From my config:
rapid_change_diff = 200
zero_offset = -10

Best regards
-xs

reinhard-brandstaedter, 2024-07-10:
You aren't running the latest version. Can you please upgrade and observe again? There have been too many changes.

reinhard-brandstaedter, 2024-07-10:
I found a potential cause for this behavior, if the contribution of the battery triggers an inverter AC output that is just around the set AC limit. In that case we'd keep the inverter limit where it is....which is fron in case it causes feed in.

xsived, 2024-07-10:
Hi Reinhard,
Now I pulled and deployed the latest SF docker image. I'll monitor the behavior of the system.
Looking at the historical data, this was the first time in about 2 months of usage, where the sf-control defected.
Great work! Thank you!
Best regards
-xs

reinhard-brandstaedter, 2024-07-11:
Yes, I believe what you observed is a rare coincidence where the hub-contribution (battery) plus an eventual contribution from panels constantly lines up at the almost exact AC limit and keeps there even though the demand. gets less.
Good catch though & thanks for reporting.

closed 2024-07-17 by reinhard-brandstaedter