reinhard-brandstaedter/solarflow-control #276: UnboundLocalError: cannot access local variable 'limit' where it is not associated with a value

joe63, 2024-07-11:
Since 2 days sf-control hangs up twice.

in the log there is following output rolling and in mqtt-explorere I don't see any telemetry values.

**UnboundLocalError: cannot access local variable 'limit' where it is not associated with a value**
2024-07-11 18:42:11,131:INFO: Triggering telemetry update: iot/A8yh63/W7798JGG/properties/read
2024-07-11 18:43:11,131:INFO: Triggering telemetry update: iot/A8yh63/W7798JGG/properties/read
2024-07-11 18:43:11,180:INFO: HUB: S:110.0W [ 109.7,109.7,110.0 ], B:100% (100), V:49.9V (49.9), C: 0W, P:True (manual, not possible), F:1.8h, E:-1.0h, H:106W, L:600W
2024-07-11 18:43:11,180:INFO: INV: AC:117.0W, DC:123.0W (63.1|59.6), L:602W (300.8W/channel) [800W]
2024-07-11 18:43:11,180:INFO: SMT: T:Smartmeter P:1021.0W [ 1021.0,1021.0 ]
2024-07-11 18:43:11,180:INFO: Direct connected panel are producing 0.0W, trying to get 1117.8W from hub.
Exception in thread Thread-441:
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
2024-07-11 18:43:11,181:INFO: Turning hub bypass OFF
    self.run()
  File "/usr/local/lib/python3.11/threading.py", line 1401, in run
    self.function(*self.args, **self.kwargs)
  File "/solarflow/utils.py", line 25, in _run
    self.function(*self.args, **self.kwargs)
  File "/solarflow/solarflow-control.py", line 422, in deviceInfo
    limitHomeInput(client)
  File "/solarflow/solarflow-control.py", line 344, in limitHomeInput
    sf_contribution = getSFPowerLimit(hub,hub_contribution_ask)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/solarflow/solarflow-control.py", line 259, in getSFPowerLimit
    log.info(f'Based on time, solarpower ({hub_solarpower:4.1f}W) minimum charge power ({MIN_CHARGE_POWER}W) and bypass state ({hub.getBypass()}), hub could contribute {limit:4.1f}W - Decision path: {path}')
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'limit' where it is not associated with a value
2024-07-11 18:44:11,132:INFO: Triggering telemetry update: iot/A8yh63/W7798JGG/properties/read
2024-07-11 18:45:11,132:INFO: Triggering telemetry update: iot/A8yh63/W7798JGG/properties/read
2024-07-11 18:45:11,180:INFO: HUB: S:110.0W [ 109.7,109.7,110.0 ], B:100% (100), V:49.9V (49.9), C: 0W, P:True (manual, not possible), F:1.8h, E:-1.0h, H:106W, L:600W
Exception in thread Thread-444:

reinhard-brandstaedter, 2024-07-12:
Yes this is due to the flaky bypass control of hub2k.
Don't use control_bypass with hub2k. it's not working properly like for hub1.2k

joe63, 2024-07-16:
@reinhard-brandstaedter 
Hallo.
I have tried the **control_bypass = false** setup for 2 days, but it's not really satisfying.  
The setup  **control_bypass = true** works good............. until to the error where the limit-value is no more filled.
When I checked the rolling error messages I restarted the sf-control-container. After restart the sf-control works as like before the not-limit-value-error.
Would it be a possibility/risk/problem to make a try /except section which sets the limit-value to a state like container start or to restart the program? eventually with a command-line-option ( restart_by_limit_error_YN)?
BR Joe

joe63, 2024-07-19:
looks good, at the moment.

Thank you!!!

tuxianerDE, 2024-08-07:
Closing issue based on your latest comment @joe63

closed 2024-08-07 by tuxianerDE