reinhard-brandstaedter/solarflow-control #285: Since last update Output to home is limited during day

AzraelsDisk, 2024-07-29:
[image]
[image]

It will switch between 1.2 and 2.1 if the value is above the 200 watt max limit.

My understanding was that "max" is used during the night... not during the day? 
Maybe I'm wrong or I don't understand the setting "discharge_during_daytime" as it is used.

Thanks
Azra

tuxianerDE, 2024-07-30:
Hey,

the max_discharge_power is always used irrespective of day and night time, as it generally limits the possible/allowed output of the Hub.

The discharge_during_daytime is a value introduced that essentially deactivates the offset and let the control script also steer the system to provide power to the house also during the day. I am personally using this as a sort of vacation mode, i.e. if nobody is home, the energy consumption is very low and thus the control script can cater the house consumption at any given point in time as I dont necessarily need to "safe" battery for the night.

tuxianerDE, 2024-10-31:
Is this issue still a topic otherwise propose and request to close otherwise I will close it after 24hrs after info.

closed 2024-11-12 by tuxianerDE