reinhard-brandstaedter/solarflow-control #290: Nachtschaltung

martin-nrw-85, 2024-09-10:
Hallo zusammen,

Mir fällt auf das der HUB1200 über Nacht garnicht mehr abschaltet und letzte Nacht dann 4% Akku verbraucht hat. Woran kann das liegen? Ich benutze das SF-Control um die Werte in einen lokalen mqtt-Brokker laufen zu lassen und in HomeAssistant läuft ein Skript zur Steuerung vom HUB-HomeOutput.
Als zusätzliche Bemerkung:
Schalte ich den Master-Switch Abends auf off, geht er nach ein paar Sekunden wieder auf on.

Ist das bei euch auch so?
Was kann man machen um den HUB über Nacht auszuschalten?

reinhard-brandstaedter, 2024-09-23:
Also sf-control macht nichts mit dem master switch. Kann ev. sein dass die telemetrie update trigger das komplette ausschalten des hubs verhindern? Schon mal versucht den hub auszuschalten wenn das sf-control nicht läuft?

Aber wenn du eh die steuerung von sf-control nicht verwendest, warum dann ueberhaupt laufen lassen? Wenn der hub eh schon an deinen lokalen MQTT reportet wozu dann das sf-control auch noch laufen lassen?

martin-nrw-85, 2024-09-29:
- Also sf-control macht nichts mit dem master switch. Kann ev. sein dass die telemetrie update trigger das komplette ausschalten des hubs verhindern?
-- kann sein

- Schon mal versucht den hub auszuschalten wenn das sf-control nicht läuft?
-- nein

- Aber wenn du eh die steuerung von sf-control nicht verwendest, warum dann ueberhaupt laufen lassen? Wenn der hub eh schon an deinen lokalen MQTT reportet wozu dann das sf-control auch noch laufen lassen?
-- Sf-controll benötige ich um im mqtt-Bereich von HA die Oberfläche mir den Schaltern zu erzeugen. Ich weiß nicht wie dies direkt in HA Konfiguriert werden kann - das lass ich mir gerne zeigen :-). 

P.S. Sorry das ich mich erst verspätet melde, Job und Freiwillige Feuerwehr lassen in meinem Alltag manchmal wenig Zeit...

closed 2025-02-21 by reinhard-brandstaedter