reinhard-brandstaedter/solarflow-control #298: src/solarflow/homeassistant

martin-nrw-85, 2024-10-17:
Hallo ich nochmal...
Wo in Home Assistant füge ich diese .json Dateien ein?

Gruß
Martin

reinhard-brandstaedter, 2024-10-17:
Das sind auto-discover templates. Wenn du die manuell hinzufuegen willst musst du die zunaechst erst mit deinen device daten ausfuellen.
Danach kannst du die via MQTT an das von deinem HA verwendeten auto-discover topic publishen.
Also zB. so:
[image]

Aber das manuell machen ist halt mau....

martin-nrw-85, 2024-10-17:
Erinnerst du dich, als du mich gefragt hast, warum ich solarflow-Control überhaupt laufen lassen, wenn ich doch eh HA die Regelung überlasse. Genau deswegen, damit mir diese Sensoren in HA gepublisht werden.

Frage: Kommt solarflow-Control auf wirklich 0 Netzeinspeisung? Mit HA komm ich auf etwa 96% Eigenverbrauch...

busaku, 2024-10-17:
Rein interessehalber, wie steuerst du das rein über HA?

Habs über HA laufen, also die Sensoren in HA, als Addon, Steuerung liegt im Solarflow-Control wie als ob man es normal laufen lässt.

Mehr als 90%-95% bekomme ich selbst unter optimalen Bedingungen nicht hin, ist aber auch sehr individuell, wenn man kocht, kann Zendure das gar nicht abdecken, da die Leistung zu hoch ist.

Hätte ich immer Akku und würde nie über Output Limit vom Akku verbrauchen denke ich würdest du gut an die 99% Selbstversorgung kommen.

martin-nrw-85, 2024-10-17:
In HA habe ich mir ein Skript dazugeschrieben geschrieben, indem ich den Solarflow Hub Output steuere, also auch ohne DTU das ganze.

reinhard-brandstaedter, 2024-10-17:
> Erinnerst du dich, als du mich gefragt hast, warum ich solarflow-Control überhaupt laufen lassen, wenn ich doch eh HA die Regelung überlasse. Genau deswegen, damit mir diese Sensoren in HA [...]

Also wenn du damit meinst keinen produzierten Strom einzuspeisen dann würde ich sagen nahezu 100%.
Wenn du dir dieses Chart (grün: Netz, gelb: Sonne, blau: Überschuss) ansiehst, gibt es zwei relevante Zeitpunkte:
- 7/2023: hier hab ich mit sf-control angefangen, mit nur 2 direkt am hub angeschlossenen Panels. Alles was produziert wurde ging entweder in den Akku oder in den Eigenverbrauch (das blaue unter der Null-Linie ist Einspeisung)
- 4/2024: hier habe ich 2 zuss. Panels aufs Dach gelegt die direkt ins Hausnetz gehen. Die sind m selben WR wie der Hub angeschlossen. Dadurch werden die Batterien schneller geladen weil weniger vom Hub für die Hausversorgung abgezweigt wird. Dadurch kommt es allerdings im Falle von SF-Bypass oder einfach zu viel Sonne auch zur Einspeisung

[image]

Jetzt im Herbst/Winter wird die Einspeisung wieder ziemlich auf 0 gehen, weil es seltener zu einem Bypass kommen wird bzw. maximal erst kurz vor Sonnenuntergang wenn nicht mehr 1600W produziert werden.

reinhard-brandstaedter, 2024-10-21:
> Rein interessehalber, wie steuerst du das rein über HA?

Das ganze Projekt fing auch mit einer reinen HA Steuerung an, kam aber dann sehr schnell an die Grenzen/Flexibilität von HA und daher eine Standalone-Lösung. Basics kann man in HA schon lösen, wenns komplizierter wird gehts anders besser.

mavo, 2024-10-24:
>Also wenn du damit meinst keinen produzierten Strom einzuspeisen dann würde ich sagen nahezu 100%.
Naja, 100% ist schwierig, im Falle von BP speist man halt ein. Ich habe aber, seit dem ich das Hub2k habe (Anfang Mai müsste das gewesen sein) ca 98% Selbstnutzung. Und wie du auch sagst, im Herbst jetzt ist 100% fast garantiert.
Übers ganze Jahr komme ich auf 97% (lt. HA energy dashboard)
[image]
Was ich aber sagen muss ist, dass SF-Control wirklich gute Arbeit leistet. Und deswegen auch hier mal wieder ein Dank an @reinhard-brandstaedter, dieses Projekt hier war der Punkt der mich überzeugt hat mir ein Speicher von Zendure zuzulegen 👍

closed 2024-11-11 by tuxianerDE