reinhard-brandstaedter/solarflow-control #300: SF Control + Zendure Cloud possible? (debugging reasons)

mavo, 2024-10-24:
Hello @reinhard-brandstaedter,

as I am facing some weird issue with my system since a while I am in an ongoing discussion with the zendure support team. The technical side asked me to connect the hub to the zendure cloud, so that they are able to debug the issue.
Is there any way to have SF-control running, whilst being connected to the zendure cloud and get the required data from their servers?
I don't want to lose days of statistics and also I still want to be able to use SF-control, as it just works very well.

The issue I am facing, just for the record:
Around a month ago, on one evening I noticed that the battery level was dropping way faster than it should be. I had the batteries at 100% that day and around 10pm the batteries did show they had "only" 70% left - where based on the usage of power it should have been more like 85%. I checked the app by connecting via BT and it was showing that only one battery was used, the other one was still at 98%.
Ever since then the battery state is all over the place for me. The problem with this comes when we are looking at min charge states, the 10% I configured is just only a number, nothing more. When I had the batteries full (such as yesterday) and I was using 3,2kWh (of the ~4kWh) it was showing its already at 10% and did shut down.
This problem amplifies itself in the days after, as charging the battery on the next day will let the system think it reaches 98% (which is not 98% in reality, it shows this very very early - usually when the total power is ~2,6kWh) and then the system will drain again in the night until it reaches 10% again. But when I am checking the actual drainage from the battery, it was able to drain more then I have charged into the batteries. This continues on and one, so that 10% shown by the batteries actually sometimes in reality is more 1%. And this is bad - especially in the winter when its cold...

Well, too much text I guess - but the important question is - is there a way to have zendure cloud + sf-control woking together? I do understand, that the update intervals might be much slower, but all is better then nothing, imo.

Best,
MaVo

reinhard-brandstaedter, 2024-10-24:
No there is no feature to use the Zendure Cloud as data-provider for sf-control.

Regarding your issue: I had a similar issue once and it was due to BMS drift. When was the last time you did a full charging/discharging cycle. going from 100% to 0% and back to 100%? IMO this should still be done on a regular basis to ensure the BMS is properly calibrated.

When talking to Zendure support for battery issues it's usually good enough to provide them charts of your battery metrics (per battery), min/max volatage in relation to eachother and the battery SoC. They don't get much more either from the hub when connected to their backend.

mavo, 2024-10-24:
Was there any solution to the issue for you?

I have this issue at least for a month now, and I did charge/discharge in this time at least 10 times. Without any improvement happening.

Zendure support for me was insisting in screenshots from their app, claiming that the data that I showed them was not from their app and therefore not valid...

How to get the charts of battery voltage etc?
Can I have this out of home assistant, or would I need to have something else to capture and visualize?

Best,
MaVo

reinhard-brandstaedter, 2024-10-24:
@mavo another thought? is your hub2k maybe already heating the battery and thus draining more (internal) power?

mavo, 2024-10-24:
> @mavo another thought? is your hub2k maybe already heating the battery and thus draining more (internal) power?

Mhhh, possible. But unlikely. It wasn't that cold when the issue started.
Also, the day when it started I connected to the hub via BT in their own app and saw this
[image]

This is not shown anymore in the app, but ever since then it is "kaputt".

tuxianerDE, 2024-10-25:
Hey @mavo

quick point to verify charge/discharge means that you actually did a full cycle i.e. up to 100% (till bypass kicks in) and down to 0%, because this appears to be a calibration topic as Reinhard had some time back.

Whilst I am afraid to ask/suggest, have to tried connecting only one battery and send it through this complete charge and discharge cycle and gather the data of it?

mavo, 2024-10-25:
> Hey @mavo
> 
> quick point to verify charge/discharge means that you actually did a full cycle i.e. up to 100% (till bypass kicks in) and down to 0%, because this appears to be a calibration topic as Reinhard had some time back.

So I am actually not sure anymore if I once more had a 100 -> 0 cycle, but I can give it a go one more time.
The cycle is 100 -> 0, or even better 100 -> 0 -> 100?

> Whilst I am afraid to ask/suggest, have to tried connecting only one battery and send it through this complete charge and discharge cycle and gather the data of it?

I did not, to be honest. I was thinking about this as well, but would also not know what I really would get out of it - other then confirmation that maybe only one battery is confused and the other one possibly fine.

tuxianerDE, 2024-10-26:
Hey,

I would suggest to let the batteries run through a complete ChargeThrough cycle, you can set the interval shorter so that the batteries will actually run through multiple i. E. 2-3 days charge through interval.

It is of course not certain that the issue disappears however with Reinhard it did and may be the BMS requires some cycle to calibrate

mavo, 2024-10-27:
So, after getting 100% yesterday and discharging it to 0 over the night it did not change anything.
I actually checked the battery percentage reported in the app and it shows stupid values even on charging.
[image]

I'll try to have custom sensors as Reinhard suggested to capture voltage of the batteries to provide this to the support.

It feels somehow broken to me right now, dunno.

Best,
MaVo

tuxianerDE, 2024-10-28:
Hey,

before you get too disappointted about it, at least when you look into the graphs of Reinhard you can see that it took 3-4 iterations until it kick back into rythm. 

Whilst it makes sense to start logging the Voltages and start the conversation with Zendure via a ticket, I would suggest to run a few more cycle (just to be safe and hopefully to still resolve it then)

mavo, 2024-10-28:
Yeah, I am currently charging again to 100% to give it another round.

>Whilst it makes sense to start logging the Voltages and start the conversation with Zendure via a ticket, I would suggest to run a few more cycle (just to be safe and hopefully to still resolve it then)

Would you be able to tell me in which MQTT topic I can find the voltages? I was not able to find it by now (@reinhard-brandstaedter, @tuxianerDE)

I have a ticket with them already, its open for a month and finally they included their technical team to it. Thats why they asked me to connect to the zendure cloud, which I actually do not want to do for various. 

@reinhard-brandstaedter another question for you. How do you visualize the data (such as you did for your problem). This somehow looks like grafana is being used. I have my own instance running now and I am able to read the data from MQTT via the MQTT data source plugin. However, this seems to only read and capture the data when I have the dashboard open. 
How do you do this?

Best,
MaVo

tuxianerDE, 2024-10-28:
Whilst the question is for Reinhard let me spoil a bit.

My setup also uses Grafana, you will have to install and configure telegraf to read your MQTT and parse it accordingly, this inserts the data into a database (my choice is influxDB) from there the Grafana Dashboard takes the data to visualize. 

Not just easy and happy to share the parsing config.
The voltages are by default not in HA. You will have to create a sensor manually, the value to look for is in the battery array called "voltage"

I can share with you as well the Json of the Grafana Dashboard, however there are dependencies to CQ's in InfluxDB so it might require some more hands on.

mavo, 2024-10-28:
Hello,

yes if you would provide me some hints on how to configure I would be a happy man.

> The voltages are by default not in HA. You will have to create a sensor manually, the value to look for is in the battery array called "voltage"

I know there are not in HA, however, I do not even see them in MQTT explorer which I have running as well.
This is what I can see as values reported:
[image]
Is it somewhere and I just did not spot it? Whats the MQTT topic to find them?

Thanks already, I really appreciate your help.

MaVo

tuxianerDE, 2024-10-28:
Hey,

short and crisp, totalVol is what you need under each battery structure.
Where you want the sample config to be posted to for telegraf?

Lg
Torsten

mavo, 2024-10-28:
Actually, wherever you want.
Can be a GitHub gist, a txt file on a web server (+ a link to it^^) or as a comment here. 
Or via PM / email on my GitHub account / email (not sure if my mail is actually public in GH)

Or maybe, if you wanna go really fancy as a post in the show and tell section in this repo so that others might be able to benefit from it as well if they want to invest some time for the setup.

As said, I am very happy to get any help.
MaVo

mavo, 2024-10-29:
Got it running so far, thank you for the input. And I actually think there is not that much polishing required.

Back to the issue: if `totalVol` is the current value - what is `maxVol` and `minVol` actually?

reinhard-brandstaedter, 2024-10-30:
MaxVol = the maximum voltage of cells in a battery
MinVol = the minimum voltage of cells in a battery

mavo, 2024-10-30:
They is not much difference, maximum offset of 1 is what I could see.

And in a very magic turn of events, zendure just send a message saying they will replace one of my batteries. I did not yet provide them any more data as I am still in the process of collecting it 🤔

The problem I have is, I cannot really say which one is potentially broken (or just not calibrated). I have a suspicion - and hope that the voltages shown on emptying the batteries can help to further strengthen this.

mavo, 2024-11-09:
As this is not a problem with SF control and Zendure has accepted to an exchange of my battery I will close this issue.
It turned out to be my first battery which had the defect as it seems. Since I have only one battery left on the hub the battery levels make sense again.

Thanks for all the inputs from all of you!

MaVo

closed 2024-11-09 by mavo