reinhard-brandstaedter/solarflow-control #302: Offset retained MQTT values overwritten at startup

tuxianerDE, 2024-10-31:
Hi,

I have observed the following behaviour, I have configured offset values in my config.ini (essentially I want t ohave them as default).

Via HA I set the offset dynamically per day to accomodate for my situation. 

Observed behaviour:
- I have a set offset values in MQTT (retained) and restart solarflow the config values are loaded at startup and then written to MQTT.

Expected behaviour (well my idea so lets discuss):
- At startup config.ini offset is loaded
- Once MQTT connection gets established and there are retained offset values on MQTT, those are getting loaded i.e. runtime update
- If there are no offset values on MQTT the values at startup are getting written to MQTT

If I am not mistaken this should be the same behaviour as for ChargeThrough or day discharge.

mavo, 2024-11-08:
I observed the same things, I guess it broke when the MQTT support for a couple of new parameters came into SF-Control.

But, whilst you mention 
> set the offset dynamically per day to accomodate for my situation

How do you do this?

I would like to see another option, which is not a offset-based version but rather a fixed time version. For me, I'd like to start discharging only at 6pm. Why you ask? Because then we are usually done with dinner prep and I do not drain the battery full steam but rather have a nice, even consumption in the house.

tuxianerDE, 2024-11-10:
Hey,

here is the part of my HA automation calculating the offset ensure that (latest) at 17.50 discharge starts. In the winter it gets darker earlier so I have a minimum value it goes to.

`metadata: {}
data:
  value: >-
    {{ ([90,((state_attr('sun.sun', 'next_setting') | as_timestamp -
    as_timestamp(today_at("17:50")))/60)|float(0)]|max|round(0)) }}
target:
  entity_id: number.solarflow_control_sunset_offset
enabled: true
action: number.set_value
`

reinhard-brandstaedter, 2024-11-20:
Fixed on dev.