reinhard-brandstaedter/solarflow-control #303: Support for Hoymiles HMS 800W-2T WiFi

oxidizer2001, 2024-10-31:
I have a Hoymiles HMS 800W-2T WiFi. For this inverter there is no OpenDTU or AhoyDTU. At the moment I use the docker image from here: https://github.com/DennisOSRM/hms-mqtt-publisher to publish the inverter data to my local mqtt brocker. It's look like this:
[image]
[image]
May be there is a way to change some in the config.ini ?
Best Regards
Rico

PS :Sorry for my bad english.

tuxianerDE, 2024-10-31:
Hi, can you set the limit of the inverter as well via the used mqtt? Because just publishing of raw data is not enough, the power of the script is in steering.

So from the above I can tell you that it is not a simple change of the ini file, sorry to disappoint

oxidizer2001, 2025-02-10:
No, i don't think so. 
I think I will go further with this project https://github.com/ohAnd/dtuGateway to bring the HMS 800W-2T work with soalrflow-control. What do you think?

tuxianerDE, 2025-02-11:
Hey,

that appears at least a good approach. What you could take a look at, since the project you linked uses also MQTT that you take a look how they publish it. It is similar to the existing format and structure of the OpenDTU you might be able to use that existing configuration.

If it is completely different we are happy to review a PR with the necessary changes that are in the DTU sub class.

oxidizer2001, 2025-02-19:
Unfortunately it is different. Here is an excerpt from the github-readme:

dtuGateway_12345678/timestamp

dtuGateway_12345678/grid/U
dtuGateway_12345678/grid/I
dtuGateway_12345678/grid/P
dtuGateway_12345678/grid/dailyEnergy
dtuGateway_12345678/grid/totalEnergy

dtuGateway_12345678/pv0/U
dtuGateway_12345678/pv0/I
dtuGateway_12345678/pv0/P
dtuGateway_12345678/pv0/dailyEnergy
dtuGateway_12345678/pv0/totalEnergy

dtuGateway_12345678/pv1/U
dtuGateway_12345678/pv1/I
dtuGateway_12345678/pv1/P
dtuGateway_12345678/pv1/dailyEnergy
dtuGateway_12345678/pv1/totalEnergy

dtuGateway_12345678/inverter/Temp
dtuGateway_12345678/inverter/PowerLimit
dtuGateway_12345678/inverter/PowerLimitSet // <-- this topic (extended with ../set) will be subscribed to get the power limit to set from your broker
dtuGateway_12345678/inverter/WifiRSSI