reinhard-brandstaedter/solarflow-control #322: Hub does not publish values to MQTT anymore

mavo, 2024-12-09:
Hello @tuxianerDE, hello @reinhard-brandstaedter,

as I mentioned in the other issue, I have quite some trouble with my setup at the moment. For some reason, the HUB does not want to send any more data into MQTT. It also fails when I try to update the Firmware of the hub whilst being connected to it via BT.
It also does not explain why it fails, it just states "update failed".

Because I have no idea anymore, I am trying to find someone who can give me some hints.

What I did yesterday was to update my NAS Firmware (bad Idea, I know but it did not want to let me wait any longer...). After this was done some of my services had all sorts of issues with networking. In the middle of it was the MQTT broker, which was not reachable for all devices anymore who previously could connect to. 
SFControl also was not able to connect to the mqtt broker anymore, although they are living in the same docker network. This issue I could resolve by exposing the MQTT container to the host network and SFControl was able to connect again. However, the hub cannot.
While I was connected to the hub via BT I saw, that the app reports that the hub is not connected to any network. BUT - as the hub was taken "offline" as most (all?!) of the SFControl users do I am not sure if this is correct what it reports.
Would one of you be able to check the settings in the app whilst being connected via BT?

Other then this, any Idea what I can try? I tried rebooting the hub, with no success.
Do you think it makes sense to run the bt-manager scripts again?

Best and thx,
MaVo

tuxianerDE, 2024-12-09:
Hey, 
short and swift. Run the bt manager again and tell it again to connect - please - to the network and thus tell him the  IP again of the mqtt, sometimes it needs a little push. I have also seen that a restart of the hub does help. 

When you run the BT manager and reconnect, make sure that you also check your MQTT with a MQTT explorer so that you at least see the connect messages.

mavo, 2024-12-09:
Mh, using the bt manager script with -i gives no valid output...

```
python3 solarflow-bt-manager.py -i
2024-12-09 16:39:24,641:INFO: scan for: zenp
2024-12-09 16:39:35,362:INFO: No Solarflow device found! You can try these steps:
                   - Move closer to the hub
                   - Reset your bluetooth connection (bluetoothctl)
                   - Restart the Solarflow Hub
                   - Disconnect any mobile Apps currently connected to the hub
```

Is this expected, when the hub was actually taken offline already? I can connect to the hub via BT from my phone (and I disconnected from it before using the manager script with -i)...
The raspi which is trying to connect is right next to the hub. This is the same raspi which I used initially to set up the hub in offline mode with the same distance (like 50cm) to the hub 🤔

tuxianerDE, 2024-12-09:
No it is not. The hub still can be used via bluetooth and this is what is being looked for. However it might happen if there is already a bluetooth device connected that this error is thrown.

1) Switch off Bluetooth on your phone! Sometimes these connections are quiet "sticky"
2) sudo systemctl restart bluetooth 
3) rerun the script.

reinhard-brandstaedter, 2024-12-10:
I’d start at the network diagnosis. Since even local docker services couldn’t connect, you moght have introduced some kind of network policies or firewalls.
Before tampering with the hub, you should make sure you can reach the mqtt broker from your local network. Ping, netcat/tcp connect, mqtt explorer from your local network…or your (isolated VLAN?) WiFi network?

mavo, 2024-12-10:
Hello,

so my broker is working again. But the hub still refuses to send any data.
The reason why I was unable to connect to the broker from one of my machines (🍎) was the latest OS update, introducing more privacy options where I did not allow docker to reach local network devices. And as MQTT explorer works fine in docker I ran it as a container on my machine. This is solved, I can connect to the broker from all devices once again.

But the issue with the hub persists. It does not send data, and neither can I find it using the BT manager o.O
I checked that it is online with my mobile after I ran the script without any success in discovery :-(

I may try to use a different raspi later today. But I am very confused why the hub does not like me anymore ^^

mavo, 2024-12-10:
Mh, something is fishy.
I now used a different raspi, still no success. I was able to update the firmware of the hub via BT connection from my phone, but also after this was done the hub is not to be found by the script on either of the raspi.
I made sure that my phone has BT off and there should not be any other device which can be connected to the hub.

I also tried to restart bluetooth service as mentioned by @tuxianerDE, also without any success.
What else can I try? The BT is clearly working on the hub, and also on the raspi as the raspi can find my phone when its BT is on and visible as BT device.

reinhard-brandstaedter, 2024-12-10:
did you set your SF_PRODUCT_ID correctly when running BT manager?
[image]

From above, you are searching for a Hub1.2k ("zenp") a Hub2k uses ("zenh")

reinhard-brandstaedter, 2024-12-10:
Well, at least it found the hub but the connection got aborted. That could be due to the reasons @tuxianerDE described (BT Stack still holding a connection, mobile app maybe as well or the hub itself).
It still doesn't detect the BT services of the hub, you should see an output like this:

```
2023-09-05 13:52:28,393:INFO: Services:
2023-09-05 13:52:28,397:INFO: 00112233-4455-6677-8899-aabbccddeeff (Handle: 18): Unknown
2023-09-05 13:52:28,402:INFO: 0000a002-0000-1000-8000-00805f9b34fb (Handle: 12): Vendor specific
2023-09-05 13:52:28,409:INFO: 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
...
```
The other parameters are not needed since -i will not try to connect to MQTT. So maybe a hub restart to clear any BT connections is useful?

mavo, 2024-12-10:
So, I restarted the hub, restarted the raspi which is trying to talk to the hub and shut down the second raspi that I had somewhat close for testing, just to be sure. My phone has BT disabled + its not even in the range. 
But still, I get the same result.

What else can I try?

mavo, 2024-12-10:
Restarted the hub another time (actually I tried to reset it, but it just did a shutdown and I restarted and now I (sometimes) get
```
python3 solarflow-bt-manager.py -i
2024-12-10 16:55:46,490:INFO: scan for: zenh
2024-12-10 16:55:47,083:INFO: Found device: 94:C9:60:AD:EE:F9: ZenH1_26
Traceback (most recent call last):
  File "solarflow-bt-manager.py", line 256, in <module>
    main(sys.argv[1:])
  File "solarflow-bt-manager.py", line 253, in main
    asyncio.run(run(broker=mqtt_broker, port=mqtt_port, info_only=info_only, connect=connect, disconnect=disconnect, ssid=ssid))
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "solarflow-bt-manager.py", line 154, in run
    async with BleakClient(device) as bt_client:
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/__init__.py", line 491, in __aenter__
    await self.connect()
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 264, in connect
    dangerous_use_bleak_cache=dangerous_use_bleak_cache
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 650, in get_services
    self._device_path, dangerous_use_bleak_cache, self._requested_services
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/manager.py", line 581, in get_services
    await self._wait_for_services_discovery(device_path)
  File "/home/pi/.local/lib/python3.7/site-packages/bleak/backends/bluezdbus/manager.py", line 686, in _wait_for_services_discovery
    raise BleakError("failed to discover services, device disconnected")
bleak.exc.BleakError: failed to discover services, device disconnected
```

reinhard-brandstaedter, 2024-12-10:
So your hub connects but immediately disconnects.
Try resetting it.
Check if your raspi is up to date, update python, update the bluetooth tools…it is Raspi 3+ i assume?

welli2009, 2024-12-10:
I have the same problem. I am connecting to the hub using my laptop. Using -i works, and I can see the data, but the hub doesn't send any data to the broker. I've installed the same update, so it seems Zendure has changed something, unfortunately.

tuxianerDE, 2024-12-10:
> [Verfolgungsbild][https://github.com/notifications/beacon/ALMQT6GSPRON7QM4NMQIECD2E42PPA5CNFSM6AAAAABTJANKNCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUW6TAS4.gif]

Sorry but what means that you have installed the same update?

welli2009, 2024-12-10:
I've connected the hub via bluetooth with my phone, installed the latest firmware update and couldn't reconnect afterwards to my local MQTT.

tuxianerDE, 2024-12-10:
What is the firmware version you are running on?

reinhard-brandstaedter, 2024-12-10:
It sound wierd, but if the disconnect worked and the hub is not reporting to zendure, it will try to local.
Maybe a few restarts. I had this behavior once and thought it is a new firmware behavior. Ot resolved itself.
Turn off the hub (don’t reset), wait, turn it on, wait, off on repeat a few times, let it run.
And while it is on, try a network sniffer on your mqtt’s network interface if you see any connection attempts.

mavo, 2024-12-10:
> […](#)

I think its 1.0.19 if I remember correct. Hub is now offline as its dark and it has no battery left (except the min charge). So I can only tell you definitely tomorrow.

mavo, 2024-12-11:
For a quick test I now connected my hub to the zendure cloud, just to see if their mqtt broker has any values for me. But, it does not.
I don't know if this is now an issue that the MQTT broker from zendure is just broken, I found some ppl saying it is on their own [repo](https://github.com/Zendure/developer-device-data-report).

Now I will try to reset the hub and take it offline again, maybe that helps. And if not I don't know. I am frustrated and I actually don't want to open another ticket for zendure which will again take months to be resolved - if it ever gets resolved.

mavo, 2024-12-11:
> When you connect the hub to zendure (reset and standard setup via the App) you should also see data reported in the App, which means it sends data via MQTT to zendure (not the developer mqtt broker - which is just a bridge to the production one).

I can see the data in the app, so it is somewhat working.

> Once you see data in the App we at least know that your hub's WiFi and setup is working...then if you get the bt-manager to disconnect it should also send to a local mqtt broker (if it is reachable from the hub).

It must be reachable from the hub, they are in the same network and the WiFi connection is apparently good enough to report data into zendure's cloud systems, so I would not know any good reason to not be able to reach my local broker.

> bluetoothctl / hcitool directly and see if that allows to connect or if you get timeouts as well.

Never had the change / need to use this directly. Followed an article on how to connect to a BT device, but was not sucessful.
- `bluetoothctl devices` shows me the zendure hub
- `bluetoothctl connect 94:C9:60:AD:EE:F9` returns 
```
Attempting to connect to 94:C9:60:AD:EE:F9
[CHG] Device D0:03:DF:F5:79:83 RSSI: -74
[CHG] Device 94:C9:60:AD:EE:F9 Connected: yes
Failed to connect: org.bluez.Error.Failed le-connection-abort-by-local
```
- `bluetoothctl pair 94:C9:60:AD:EE:F9`
```
Attempting to pair with 94:C9:60:AD:EE:F9
[CHG] Device 94:C9:60:AD:EE:F9 Connected: yes
Failed to pair: org.bluez.Error.ConnectionAttemptFailed
```

Am I doing anything wrong? Could you help me out here maybe?

mavo, 2024-12-11:
> However if the disconnect with bt-manger works once and is successful you should have data in MQTT at these topics:
iot/<SF_PRODUCTID>/<SF_DEVICE_ID>/register/replay
(that topic is published from bt-manager during the disconnect process and also confirms that your MQTT is reachable from bt-manager)

I restarted the hub once more and I can see in MQTT explorer, that the value of the topic did change. I manually change the last retained msg in the broker, so that I can see if it changes. And it does.
But still, nothing is reported afterwards 🤔 
[image]

Further ideas? :-(

reinhard-brandstaedter, 2024-12-11:
a full tcpdump on the broker side network interface on your MQTT's port.

mavo, 2024-12-11:
Can you help a helpless man.
Would it be enough to run this on the docker container when exec into it?

Or does it need to be on the host system - which is my NAS - where I am not even sure if I can install tcpdump or if it would not just be a lot of output as it is used as my main gateway for all devices in the network?

mavo, 2024-12-11:
Seems like my NAS does not want to have tcpdump on it. I'll try tomorrow to remap the hub to another broker (quickly set up one) on one of my raspis and check if a) it reports data and if not b) run tcpdump. 
I used it now for a bit and I can see connections between my mqtt explorer and the broker, so it works to capture stuff. I'll let you know / send you output, as I probably will need some assistance to decypher it.

reinhard-brandstaedter, 2024-12-12:
Just to be sure...your hub still has enough battery (> than the minSoc is set)? Because if it is lower it will not report anything...

mavo, 2024-12-12:
Oh boi... rly? Even if the "sun is out" (yes, its winter, do not know what sun actually looks like atm)?
This would not really make sense, as I used it quite some times already to not use CHARGE_THROUGH but instead just set the minSoc to 100% manually. And then the hub did indeed report its input data such as the solar power and battery change values 🤔 
Yes, it does not report anything when the sun is down, but on daytime it should... Should it?!

reinhard-brandstaedter, 2024-12-12:
Well the hub could shut itself completely off…but if you got data to the cloud its maybe not the case.

mavo, 2024-12-12:
Yeah ok, but in general it should send data. Which it does not.
I now set the hub to a lower minSoc than its current charge, I can see in the dtu it's ejecting power into the inverter.

But there is still no data reported. Today I also tried using a different broker on one of the raspis, same behavior, the initial message appears but nothing else.

Maybe the new firmware really did break something and it refuses to send any data locally.
I also had tcp dump running on the pi, which did not show anything except my connection from mqtt explorer. Unfortunately, the tcpdump was not running when the initial message was sent. But I'll reconfigure the hub once more later today to see if I can at least capture this.

As I can see values via BT being sent when running the bt-manager with -i I am considering using this for a while. I know, you do not recommend this, but there is something super off for me atm.
A question regarding the BT manager as "proxy", will the BT connection restore itself when the hub was offline in the night, or would this be sth I had to do manually?

mavo, 2024-12-12:
So, I now flipped it to use the raspis broker once again and had tcpdump running. I can see that one time there was a connection from the raspi's IP to the broker, but afterwards - silence. The only thing thats going on is the ping pong between my machine and the broker because mqtt explorer is running :-(

I now also tried running the bt manager script - and its printing some odd things:
```
2024-12-12 21:11:25,155:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'packInputPower': 35, 'outputHomePower': 34}}
2024-12-12 21:11:25,375:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 0}}
2024-12-12 21:11:25,974:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 0}}
2024-12-12 21:11:26,195:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 1}, 'packData': [{'maxTemp': 2771, 'sn': 'CO4HMEFMF701354'}]}
2024-12-12 21:11:26,414:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 0}}
2024-12-12 21:11:27,215:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'packInputPower': 34, 'outputHomePower': 33}}
2024-12-12 21:11:27,814:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 1}}
2024-12-12 21:11:28,054:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 0}}
2024-12-12 21:11:28,195:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {}, 'packData': [{'maxTemp': 2781, 'sn': 'CO4HMEFMF701354'}]}
2024-12-12 21:11:29,034:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 1}}
2024-12-12 21:11:29,255:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 0, 'packInputPower': 33, 'outputHomePower': 32}}
2024-12-12 21:11:29,654:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 1}}
2024-12-12 21:11:29,854:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 0}}
2024-12-12 21:11:31,074:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 1}}
2024-12-12 21:11:31,295:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'electricLevel': 39, 'wifiState': 0, 'packInputPower': 32}}
2024-12-12 21:11:31,455:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'outputHomePower': 31, 'packInputPowerCycle': 21, 'outputHomePowerCycle': 20}}
2024-12-12 21:11:31,715:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {}, 'packData': [{'power': 14, 'state': 2, 'sn': 'CO4HLHALKP05056'}]}
2024-12-12 21:11:31,895:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 1}, 'packData': [{'power': 14, 'state': 2, 'sn': 'CO4HMEFMF701354'}]}
2024-12-12 21:11:32,335:INFO: {'method': 'report', 'deviceId': 'ETFTke36', 'properties': {'wifiState': 0, 'packInputPower': 31, 'outputHomePower': 30}}
```

Notice the wifi state 1 and 0 logs all over the place O.o
This is super strange 🤔

reinhard-brandstaedter, 2024-12-13:
Ok this is indeed strange (the flipping of wifi on/off). But this would explain a lot!
Is your wifi WPA3? Did you ever reset your AccessPoint? Try to change the WiFi (eg add a temporary less secure/open WiFi network) and join that one?

mavo, 2024-12-13:
> Is your wifi WPA3?

No, WPA2.

> Did you ever reset your AccessPoint?

Restarted all AP's and the router a couple of times, yes.

> Try to change the WiFi (eg add a temporary less secure/open WiFi network) and join that one?

Lets see if I can momentarily use my phones hotspot to try that.

mavo, 2024-12-13:
Mh, tested using my phone as hotspot - with the result that the phone does not show any device connected using its hotspot + wifi 1/0 switching happening all the time.

Do you perhaps know how the hub identifies itself to the router, so that I can check why it might be unable to connect? I am using a fritzbox with a couple of fritzrepeaters in a mesh wifi. And this is actually unchanged for a loooong time.

The strange thing is, I can set up the network via the app and then its working, as I can see data in the zendure cloud. So it cannot really be an issue with the wifi in general, would you agree to this assessment?

mavo, 2024-12-13:
Well, now I also found an older router I had in a shelf. Created own wifi from this and connected it via LAN into my default network. Same thing happening, wifi mode still switches on and off.

I now went back to the original network to see if I can find it in my router. 
Based on the MAC the bt manager scripts print when running, I think I found the corresponding device (BT mac seems to be +1 to wifi mac). And it says its connected and active - with a stable connection as it seems O.o

mavo, 2024-12-13:
Just to make sure I am not missing anything and just being very stupid.
In the BT manager script, there are 5 (think it's 5...?) environment variables that need to be exported.
There is, besides the SF IDs and my wifi pwd the mqtt user and mqtt pwd in. As the broker needs to allow anonymous logins the values in those variables are not relevant. Or are they - and if they are, what do I need to set? My broker literally has no users, all clients are anonymous.

reinhard-brandstaedter, 2024-12-14:
if your mqtt doesn’t need authentication then user/pwd are not relevant.
But the hub will always send a user/pwd as this os hardcoded.
The MQTT protocol also needs a clientId which is also hardcoded in the hub. A broker will not accept twice the same client ID. So iyour broker should always clean and remove old connections, which is the default behavior I guess. Maybe to be sure check mqtt logs on debug level?

mrostock, 2024-12-14:
I am following your discussion here and maybe I can give you a hint to fix your connection problem.

As I tried to disconnect my hub from cloud and connect it to my own mqtt, I had an issue with the wifi password. Due to a special character in my wifi password, the password value (environment variable) was cut of. So the password was wrong.

To fix this, I set my password directly in code and left the environment variable empty.

May this can help?!?

mavo, 2024-12-14:
> But the hub will always send a user/pwd as this os hardcoded.

Oh boi 🙈 this might be the issue. As I had issues with my broker after the updates I made I changed a couple of things. And I now also remember one change I made was adding a password file to the mqtt conf. This was left blank, as I had annonymous logins enabled and did not think about it much more -.-
And yes... I could not connect to my broker with a random username / password (just tested). So I changed it and now this works again... 
 somehow expect it to suddenly work again when the hub gets enough energy from the "sun".

And if this is the case, I will burry myself and never be seen again 🙈

mavo, 2024-12-15:
Well... I am an idiot. Sorry for taking your time.

closed 2024-12-15 by mavo

reinhard-brandstaedter, 2024-12-15:
Well those things happen, glad you resolved it @mavo