reinhard-brandstaedter/solarflow-control #324: Version 0.76 not Starting

sturowski, 2024-12-13:
Version 0.76 is not starting herere is the Traceback:
```bash
sf-control | Traceback (most recent call last):
sf-control |   File "/solarflow/solarflow-control.py", line 595, in <module>
sf-control |     main(sys.argv[1:])
sf-control |   File "/solarflow/solarflow-control.py", line 592, in main
sf-control |     run()
sf-control |   File "/solarflow/solarflow-control.py", line 525, in run
sf-control |     client.loop_forever()
sf-control |   File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1756, in loop_forever
sf-control |     rc = self._loop(timeout)
sf-control |          ^^^^^^^^^^^^^^^^^^^
sf-control |   File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1164, in _loop
sf-control |     rc = self.loop_read()
sf-control |          ^^^^^^^^^^^^^^^^
sf-control |   File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1556, in loop_read
sf-control |     rc = self._packet_read()
sf-control |          ^^^^^^^^^^^^^^^^^^^
sf-control |   File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
sf-control |     rc = self._packet_handle()
sf-control |          ^^^^^^^^^^^^^^^^^^^^^
sf-control |   File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3033, in _packet_handle
sf-control |     return self._handle_publish()
sf-control |            ^^^^^^^^^^^^^^^^^^^^^^
sf-control |   File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3327, in _handle_publish
sf-control |     self._handle_on_message(message)
sf-control |   File "/opt/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
sf-control |     on_message(self, self._userdata, message)
sf-control |   File "/solarflow/solarflow-control.py", line 138, in on_message
sf-control |     smartmeter.handleMsg(msg)
sf-control |   File "/solarflow/smartmeters.py", line 105, in handleMsg
sf-control |     self.updPower()
sf-control |   File "/solarflow/smartmeters.py", line 80, in updPower
sf-control |     log.info(f'SMT triggers limit function: {previous} -> {self.getPower()}: {"executed" if self.trigger_callback(self.client,force=force_trigger) else "skipped"}')
sf-control |                                                                                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sf-control |   File "/solarflow/solarflow-control.py", line 471, in limit_callback
sf-control |     limitHomeInput(client)
sf-control |   File "/solarflow/solarflow-control.py", line 374, in limitHomeInput
sf-control |     sf_contribution = getSFPowerLimit(hub,hub_contribution_ask)
sf-control |                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sf-control |   File "/solarflow/solarflow-control.py", line 237, in getSFPowerLimit
sf-control |     sunrise_off = timedelta(minutes = SUNRISE_OFFSET)
sf-control |                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sf-control | TypeError: unsupported type for timedelta minutes component: NoneType
sf-control | 2024-12-13 20:33:37,731:INFO: Triggering telemetry update: iot/A8yh63/5W9Pw17L/properties/read
sf-control | 2024-12-13 20:34:37,733:INFO: Triggering telemetry update: iot/A8yh63/5W9Pw17L/properties/read
sf-control | 2024-12-13 20:34:37,871:INFO: HUB: S:0.0W [ 0.0 ], B: 1% ( 1) low: 10 high: 100, V:49.1V (49.1), C: 0W, P:False (manual, possible), F:249.7h, E:-1.0h, CT:OFF (32hrs) idle, H: 0W, L:400W
sf-control | 2024-12-13 20:34:37,871:INFO: INV: AC:88.4W, DC:93.0W (0.0|0.0|46.3|46.7), L:607W (151.7W/channel) [ -1W]
sf-control | 2024-12-13 20:34:37,872:INFO: SMT: T:Smartmeter P:2387.0W [ 2387.0,2387.0 ]
sf-control | 2024-12-13 20:34:37,872:INFO: Direct connected panels (88.4W) can't cover demand (2455.4W), trying to get 2367.0W from hub.
sf-control | Exception in thread Thread-3:
sf-control | Traceback (most recent call last):
sf-control |   File "/usr/local/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
sf-control |     self.run()
sf-control |   File "/usr/local/lib/python3.11/threading.py", line 1401, in run
sf-control |     self.function(*self.args, **self.kwargs)
sf-control |   File "/solarflow/utils.py", line 25, in _run
sf-control |     self.function(*self.args, **self.kwargs)
sf-control |   File "/solarflow/solarflow-control.py", line 475, in deviceInfo
sf-control |     limitHomeInput(client)
sf-control |   File "/solarflow/solarflow-control.py", line 374, in limitHomeInput
sf-control |     sf_contribution = getSFPowerLimit(hub,hub_contribution_ask)
sf-control |                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sf-control |   File "/solarflow/solarflow-control.py", line 237, in getSFPowerLimit
sf-control |     sunrise_off = timedelta(minutes = SUNRISE_OFFSET)
sf-control |                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sf-control | TypeError: unsupported type for timedelta minutes component: NoneType
```
I think the problem is that in src/solarflow/solarflow-control.py you change a lot of the config loading to None like this:
```python
# flag, which can be set to allow discharging the battery during daytime
DISCHARGE_DURING_DAYTIME = None   #config.getboolean('control', 'discharge_during_daytime', fallback=None) or bool(os.environ.get('DISCHARGE_DURING_DAYTIME',False))

#Adjustments possible to sunrise and sunset offset                                                                                        
SUNRISE_OFFSET = None   #config.getint('control', 'sunrise_offset', fallback=60) or int(os.environ.get('SUNRISE_OFFSET',60))                                               
SUNSET_OFFSET = None    #config.getint('control', 'sunset_offset', fallback=60) or int(os.environ.get('SUNSET_OFFSET',60))   
```

Why do you changed it or I'm using the config wrong? I updated from version 0.69 to 0.76.
King regards & Thanks
Sven

reinhard-brandstaedter, 2024-12-14:
Try the master tagged package from ghcr.io. (in the packages section, not the releases section)
I think something we t wrong when moving to github actions builds.

The config change now allows on-the-fly updates via MQTT.

sturowski, 2024-12-18:
Thanks for the fix!

closed 2024-12-18 by sturowski