reinhard-brandstaedter/solarflow-control #331: Type Error when subtracting "None" from "int"

DB6CP, 2024-12-19:
Opened a new issue since this could go missing when burried in the Home Assistant Addon Issue 

TypeError: unsupported operand type(s) for -: 'int' and 'NoneType'
[09:51:49] INFO: Running Solarflow Control
Traceback (most recent call last):
File "/solarflow/solarflow-control.py", line 82, in
or int(os.environ.get('MAX_INVERTER_INPUT',MAX_INVERTER_LIMIT - MIN_CHARGE_POWER))
~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~

Seems like MIN_CHARGE_POWER is not set or it is not recognized properly.

reinhard-brandstaedter, 2024-12-19:
do you have min_charge_power set in your config.ini?
```
[control]
min_charge_power = 125
```

DB6CP, 2024-12-19:
Config Ini should be filled from "Konfigurationen" in Home Assistant. So yes it should be set.
In The Config.yaml 125 is also set as default for min_charge_power

reinhard-brandstaedter, 2024-12-20:
max_inverter_input is the value that has to be set in the config properly. This MUST be set, because this is the mandatory value that you want to feed to the inverter (to not blow it up).

Agree the fallback is now not type safe since min_charge_power is now late-initialized:
```
MAX_INVERTER_INPUT =    config.getint('control', 'max_inverter_input', fallback=None) \
                        or int(os.environ.get('MAX_INVERTER_INPUT',MAX_INVERTER_LIMIT - MIN_CHARGE_POWER))
```

But if max_inverter_input is set properly this fallback (MAX_INVERTER_LIMIT - MIN_CHARGE_POWER) would never be executed. Maybe I just set this to a hardcoded default of 400W (something that every inverter should be able to handle on a single channel...

DB6CP, 2024-12-20:
ok so creating another Docker container and not using the HA-Addon now it seems to run. ( had also problems mounting the config but that was a Layer 9 Problem on my side ) seems to work now.

closed 2024-12-22 by reinhard-brandstaedter