reinhard-brandstaedter/solarflow-control #332: Hub wirds als Panel verarbeitet

DB6CP, 2024-12-21:
So nun kurz in deutsch:

Ich hab jetzt endlich ein Funktionierendes System. Hub200 / AB2000 das ganze an einem HMS800-2T mit OpenDtu und als Smartmeter ein ShellyEM3. Panels sind nur an den Hub angeschlossen, der Hoymiles bekommt nur den Hub und kein eigenes Panel. Solar Flow Control läuft als Addon im Homeassistant. 

Gestern Nacht habe ich dann mal die Akku Restladung ins Haus entladen. Dabei ist mir aufgefallen das die Leistung die von ganzen System abgerufen wird ziemlich stark schwankt. Ich hatte zu dem Zeitpunkt das Auto am laden ( Also eindeutig mehr als der Hub + Hoymiles liefern kann ). Ich hätte erwartet das das ganze System dann auf Maximale Leistung ( Bei mir momentan auf 600W eingestellt ) setzt und dann loslegt. Gesehen habe ich aber schwankungen zwischen 200W und 600W. 
[image]

Im Log selbst sehe ich das anscheinend die Leistung die ich aus dem Hub an den Wechselrichter schicke als Panel Leistung interpretiert wird. 
[image]

Muss ich in meiner Config noch irgendwas umstellen das er merkt das ich keine direkt angeschlossenen Panels am Wechselrichter habe? 
Kann es sein das hier der MPPT vom Hoymiles sich mit dem Ausgang des Hub2000 in die Haare bekommt ? 

Ich hoffe das ist jetzt keine zu Banale Frage.

reinhard-brandstaedter, 2024-12-21:
> Kann es sein das hier der MPPT vom Hoymiles sich mit dem Ausgang des Hub2000 in die Haare bekommt ?

Das schaut ziemlich danach aus. Hast du ein V/A diagram des Hoymiles ev? Ich nehm an du verwendest die y-Kabel?

DB6CP, 2024-12-21:
Genau Kanäle sind auf 3 eingestellt. V/A Diagramm hab ich jetzt keins.

DB6CP, 2024-12-21:
Zum Thema VA, also ich hätte ja im MQTT Broker vom Hoymiles die Kanäle 1 und 2. Aber leider sind die nicht als Sensor im HomeAssistant verfügbar ( Sonst hätte ich über die Verlaufsdaten ja Spannung und Strom ). Da ich aber erst frisch mit der ganzen Home Assistan Sache anfange ist das anlegen eines Sensors für HA noch ein wenig ein Buch mit 7 Siegeln.

reinhard-brandstaedter, 2024-12-21:
> Genau Kanäle sind auf 3 eingestellt. V/A Diagramm hab ich jetzt keins. 

Der HMS800 hat zwei eingänge, also solltest du das auf [1,2] oder noch besser uf [1] oder [2] stellen. Je nachdem auf welchem Anschluss eben der Hub hängt.

> Es sind mehrere Y Kabel in verwendung. Ich hab 2 Panels einer ausrichtung auf ein Y auf einen Eingang des Hubs, die anderen beiden auf den anderen eingang. Gegenüber dem Hoymiles ist der Ausgang des Hub per Y aufgeteilt auf die beiden Eingänge

Ich würde an deiner Stelle eines der 4 panels direkt an den HMS hängen, das y-kabel zwischen hub und WR auf jeden Fall weg. Oder vom Hub nur einen Anschluss des WR bedienen (auf max eingang achten)

DB6CP, 2024-12-21:
hmm. Ich hatte bisher einen Growatt Wechselrichter. Der hing genau in derselben Config am Hub. Das hat geklappt. Leider war der Growatt nicht ordentlich steuerbar. Ich hatte eigentlich gehofft das ich da mit dem Hoymiles und der OpenDTU besser hinbekomme. Ich hab jetzt mal auf [1,2] angepasst. Wenns garnicht klappt dann wird ein Anschluss abgeklemmt. Ich berichte. 

Danke für die Hilfe, auch wenn das jetzt weniger mit der Steuerung als mehr mit der Hardware zu tun hat.

DB6CP, 2024-12-21:
Also alleine das anpassen auf [1,2] schaut im Log jetzt gut aus. 
Hub läuft Gerade auf Max. Also 700W ins Haus/Hoymiles und 200W in den Akku. 
[image]

So hatte ich mir das auch vorgestellt.

reinhard-brandstaedter, 2024-12-21:
> hmm. Ich hatte bisher einen Growatt Wechselrichter. Der hing genau in derselben Config am Hub. Das hat geklappt. Leider war der Growatt nicht ordentlich steuerbar. Ich hatte eigentlich gehofft das [...]

Es kann eh funktionieren mit den y-kabel von 1 Hub Ausgang auf 2 Inverter Eingänge. Bei manchen Spannungs/Stromkombinationen wurde aber schon von vielen beobachtet dass sich dann eben die 2 MPPT gegenseitig "stören" und dann kommt man in solche Schwingungsszenarien (und einhergehend mit einer schlechten Leistungsbilanz).
Würde ich genau beobachten in deinem Fall. Das Verhalten wurde übrigens auch von Hoymiles selbst so bestätigt.

DB6CP, 2024-12-21:
Weiteren Fehler gefunden. Shelly Klemmen waren nicht genau da wo sie hingehören. Nach Korrektur klappt jetzt alles. Ich sehe 
Inverterkennlinie schaut auch soweit gut aus. Auch bei veränderten Leistungswerten passiert jetzt kein Aufschwingen mehr:
[image]

Scheint also mit Y auch mit Hoymiles zu funktionieren. 

Aber ein paar graue Haare hab ich jetzt schon mehr.

closed 2025-01-08 by reinhard-brandstaedter