reinhard-brandstaedter/solarflow-control #336: full_charge_interval = 120

joe63, 2024-12-29:
Welche Einstellung benötige ich damit der full_charge_interval = 120, wert schlagen wird. Mein zendure ab2000 wird seit Wochen nicht mehr voll geladen. 
[image]
[image]

Was sagt der dry-run aus?

Danke 
Lg Joe

reinhard-brandstaedter, 2024-12-29:
Eigentlich keine, wenn der hub nach 5 tagen(120h) nicht wieder mal voll war wird er nicht mehr entladen bis er voll ist.
Dry run ist nun mal dry run…es wird nicht wirkluch gesteuert (zum testen).

joe63, 2024-12-29:
Hallo,
meine Einstellung  **full_charge_interval = 120** funktioniert irgend wie nicht! Die Batterie wurde schon länger nicht mehr geladen. (könnte es sein, weil ich den Discharge Limit auf 15% gestellt habe ( um die Tiefentladungen zu verhindern).

anbei der akkulevel für die letzten 30 Tage
[image]

heute um ca. 15:37 ( im log 14:37) wurde wieder begonnen die Batterie zu entladen. 
[image]

anbei das logfile
[10000_log.txt](https://github.com/user-attachments/files/18270206/10000_log.txt)

reinhard-brandstaedter, 2024-12-30:
Welche version läuft bei dir?
Config.ini Inhalt?
Check mal im mqtt topic ob chargeThrough auch tatsächlich ON ist [Siehe hier](https://github.com/reinhard-brandstaedter/solarflow-control/tree/master/doc)

joe63, 2024-12-30:
Welche version läuft bei dir? **wo sehe ich das? kann man die Version abfragen?** 

Config.ini Inhalt? **siehe Anhang**

Check mal im mqtt topic ob chargeThrough auch tatsächlich ON ist. **
[config_anonym_ini.txt](https://github.com/user-attachments/files/18273261/config_anonym_ini.txt)
**
[image]

reinhard-brandstaedter, 2024-12-30:
Bei dir fehlen in der config:
control_soc = true
battery_high = xx (98)
battery_low = xx (15)

Vom Image has her wuerd ich sagen du verwendest eine version aus dem Juli. Du solltest ev. mal das image neu pullen z.B. von [hier](https://github.com/reinhard-brandstaedter/solarflow-control/pkgs/container/solarflow-control/329134645?tag=master)

joe63, 2024-12-30:
Hi,
diese 3 Parameter habe ich meistens nach einem server-restart im HomeAssistant verändert, werde ch aber sofort in die config.ini schreiben.

am ende im [control]-abschnitt ist ok?

werde auch ein aktuelles image pullen. hoffe es ist genauso zu starten wie das juli image.

joe63, 2024-12-30:
erhalte diesen Error beim Start:

joe@joe-ESPRIMO-Q920:/usr/local/sbin/sfcontrol$ sudo docker run -v ${PWD}/src/config.ini:/solarflow/config.ini --restart=always --name solarflow-control ghcr.io/reinhard-brandstaedter/solarflow-control:master
Traceback (most recent call last):
  File "/solarflow/solarflow-control.py", line 82, in <module>
    or int(os.environ.get('MAX_INVERTER_INPUT',MAX_INVERTER_LIMIT - MIN_CHARGE_POWER))
                                               ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for -: 'int' and 'NoneType'
joe@joe-ESPRIMO-Q920:/usr/local/sbin/sfcontrol$

reinhard-brandstaedter, 2024-12-30:
Dir fehlt auch max_inverter_input in der config.ini
https://github.com/reinhard-brandstaedter/solarflow-control/blob/95f0132f5e87047048fc25b6495bfb542fad22ff/src/config.ini#L102

joe63, 2024-12-30:
Danke, sieht jetzt gut aus!

(derzeit) letzte Unklarheit:
ich habe als battery-max-level 98% eingestellt, sehe den Wert zwar im log von sf-control, nicht aber in der statuspage und auch nicht im HA.

**config.ini**
battery_high = 98
battery_low = 15

**sf-control-log**
2024-12-30 14:26:33,266:INFO: Updating charge through stage: idle => charging
2024-12-30 14:26:33,267:INFO: Reading last empty time: 2024-11-25 07:40:14
2024-12-30 14:26:33,267:INFO: Updating full charge interval to 120hrs
2024-12-30 14:26:33,268:INFO: Solarflow set DryRun: False
2024-12-30 14:26:33,268:INFO: OpenDTU set DryRun: False
2024-12-30 14:26:33,347:INFO: Received charge-through control: ON, Control SoC: False, SocMax: 98%, SocMin: 15%
2024-12-30 14:26:36,542:INFO: Updating charge through stage: charging => discharging

**sf-statuspage**
[image]

**HA**
[image]

reinhard-brandstaedter, 2024-12-30:
control_soc in der config.ini ist wohl noch auf false statt auf true?
Aber das würde bewirken dass er eben die limits beim charge through auf 0/100 setzt und danach wieder auf deine konfigurierten 15/98 damit auch tatsächlich ein voller zyklus passiert.

joe63, 2024-12-30:
eigentlich schon
[image]

ich sehe mir das mal ein paar tage an und melde mich dann wieder.

Danke für das tolle Projekt und den guten Support. 
Guten Rutsch und viel Glück für das 2025er Jahr.
lg Joe

reinhard-brandstaedter, 2025-01-11:
Aenerdungen in der src/config.ini bringen dir gar nix, das ist nur das template file. Die "echte" config.ini liegt im verzeichnis in dem das script ausgeführt wird.
Bzw. nachdem deine config variablen ja auch im MQTT als retained hinterlegt sind werden die genommen:

```
2025-01-11 11:14:25,344:INFO: Note: Solarflow Control persists initial configuration settings in your MQTT broker and will use those first (if found) to allow on-the-fly updates!
2025-01-11 11:14:25,344:INFO: If you want to override these values from your config.ini you need to clear those retained topics in your broker first!
2025-01-11 11:14:25,345:INFO: Connected to MQTT Broker!
2025-01-11 11:14:25,345:INFO: Found control/sunriseOffset, set SUNRISE_OFFSET to 30 minutes
2025-01-11 11:14:25,345:INFO: Found control/sunsetOffset, set SUNSET_OFFSET to 30 minutes
2025-01-11 11:14:25,345:INFO: Found control/minChargePower, set MIN_CHARGE_POWER to 100W
2025-01-11 11:14:25,345:INFO: Found control/maxDiscahrgePiwer, set MAX_DISCHARGE_POWER to 210W
2025-01-11 11:14:25,345:INFO: Found control/dischargeDuringDaytime, set DISCHARGE_DURING_DAYTIME to False
2025-01-11 11:14:25,346:INFO: Found control/batteryTargetSoCMin, set BATTERY_LOW to 15%
2025-01-11 11:14:25,346:INFO: Found control/batteryTargetSoCMax, set BATTERY_HIGH to 98%
2025-01-11 11:14:35,346:INFO: Control Parameters:
2025-01-11 11:14:35,346:INFO:   MIN_CHARGE_POWER = 100
2025-01-11 11:14:35,347:INFO:   MAX_DISCHARGE_LEVEL = 210
2025-01-11 11:14:35,347:INFO:   MAX_INVERTER_LIMIT = 800
2025-01-11 11:14:35,347:INFO:   MAX_INVERTER_INPUT = 800
2025-01-11 11:14:35,347:INFO:   SUNRISE_OFFSET = 30
2025-01-11 11:14:35,347:INFO:   SUNSET_OFFSET = 30
2025-01-11 11:14:35,347:INFO:   BATTERY_LOW = 15
2025-01-11 11:14:35,347:INFO:   BATTERY_HIGH = 98
2025-01-11 11:14:35,347:INFO:   BATTERY_DISCHARGE_START = 10
2025-01-11 11:14:35,347:INFO:   DISCHARGE_DURING_DAYTIME = False
```

hier auch korrektes Verhalten. last full time ist laenger her als 120h => also wird das max level auf 100% angehoben damit mal voll geladen wird. charge through stage: idle => charging

```
2025-01-11 11:14:35,418:INFO: Reading last full time: 2025-01-03 14:44:23
2025-01-11 11:14:35,418:INFO: Reading battery target mode: discharging
2025-01-11 11:14:35,418:INFO: Solarflow set DryRun: False
2025-01-11 11:14:35,418:INFO: OpenDTU set DryRun: False
2025-01-11 11:14:38,680:INFO: Received charge-through control: True, Control SoC: True, SocMax: 98%, SocMin: 15%
2025-01-11 11:14:38,680:INFO: To turn on charge-through we need to adjust the max SoC from 98% to 100%!
2025-01-11 11:14:38,680:INFO: Setting maximum charge level to 100%
2025-01-11 11:14:38,680:INFO: To turn on charge-through with a full-cycle we need to adjust the min SoC from 15% to 0%!
2025-01-11 11:14:38,680:INFO: Setting minimum charge level to 0%
2025-01-11 11:14:38,680:INFO: Set ChargeThrough: False => True
2025-01-11 11:14:38,680:INFO: Updating charge through stage: idle => charging
2025-01-11 11:14:38,680:INFO: Setting maximum charge level to 100%
2025-01-11 11:14:38,680:INFO: Setting minimum charge level to 15%
2025-01-11 11:14:38,763:INFO: Received charge-through control: ON, Control SoC: True, SocMax: 98%, SocMin: 15%
2025-01-11 11:14:52,209:INFO: Determined inverter's max capacity: 800.0
2025-01-11 11:15:35,348:INFO: Triggering telemetry update: iot/A8yh63/W7978JGG/properties/read
2025-01-11 11:16:35,349:INFO: Triggering telemetry update: iot/A8yh63/W7978JGG/properties/read
2025-01-11 11:16:35,411:INFO: [31;20mHUB: S:57.5W [ 57.5,57.5 ], B: 28% (28) low: 15 high: 100, V:49.7V (49.7), C: 56W, P:False (manual, possible), F:188.5h, E:1131.6h, CT:ON (120hrs) charging, H: 0W, L: 0W[0m
2025-01-11 11:16:35,412:INFO: [33;20mINV: AC:0.0W, DC:0.2W (0.1|0.1), L: 20W (10.0W/channel) [800W][0m
```

Der Hub uebernimmt die settings auch:
low: ist noch auf 15% (wird erst auf 0 gesetzt wenn die 100% erreicht wurde
```
2025-01-11 11:16:35,411:INFO: [31;20mHUB: S:57.5W [ 57.5,57.5 ], B: 28% (28) low: 15 high: 100, V:49.7V (49.7), C: 56W, P:False (manual, possible), F:188.5h, E:1131.6h, CT:ON (120hrs) charging, H: 0W, L: 0W[0m
2025-01-11 11:16:35,412:INFO: [33;20mINV: AC:0.0W, DC:0.2W (0.1|0.1), L: 20W (10.0W/channel) [800W][0m
2025-01-11 11:16:35,412:INFO: [33;32mSMT: T:Smartmeter P:280.0W [ 282.8,282.8,283.7,284.1,284.7,284.8 ][0m
2025-01-11 11:16:35,412:INFO: Direct connected panel are producing 0.0W, trying to get 264.8W from hub.
2025-01-11 11:16:35,413:INFO: Based on time, solarpower (57.5W) minimum charge power (100W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2025-01-11 11:16:35,413:INFO: Charge-Through is active! To ensure it is fully charged at least every 120hrs not discharging now!
```

Also wenn du da nicht via HA in die settings reinpfuschst, die topics in deinem MQTT broker zwischendurch ausräumst oder sonst irgendwas umstellst oder diconnectest sollte der das ganz normal bis auf 100% raufladen und dann runter auf 0%.
Hat er ja auch schon gemacht wie ich in den charts sehe.

Lass mal am besten "nativ" laufen und nicht als HA addon um andere fehlerquellen auszuschliessen.

joe63, 2025-01-11:
lt. kommandoaufruf erwartet sf-control im src-pfad, wenn sie im aufrufpfad steht dann error(im src-pfad ohne error)

joe@joe-ESPRIMO-Q920:/usr/local/sbin/sfcontrol$ sudo nano config.ini
joe@joe-ESPRIMO-Q920:**/usr/local/sbin/sfcontrol$ sudo docker run -v ${PWD}/src/config.ini:/solarflow/config.ini --restart=always --name solarflow-control ghcr.io/reinhard-brandstaedter/solarflow-control:master**
2025-01-11 16:37:19,885:ERROR: No configuration file (config.ini) found in execution directory! Using environment variables.
Traceback (most recent call last):
  File "/solarflow/solarflow-control.py", line 82, in <module>
    or int(os.environ.get('MAX_INVERTER_INPUT',MAX_INVERTER_LIMIT - MIN_CHARGE_POWER))
                                               ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for -: 'int' and 'NoneType'
joe@joe-ESPRIMO-Q920:/usr/local/sbin/sfcontrol$ ls
config.ini  src
joe@joe-ESPRIMO-Q920:/usr/local/sbin/sfcontrol$

-------------------------------------------------------------

[image]

[image]

habe den HA derzeit pausiert, Nachteil: andere Werte z.B Gasverbrauch wird derzeit nicht gemessen.  werde mir das mal ansehen, bis zumindest der Akku wieder voll ist und dann den HA wieder aktivieren. und eine logdatei sichern....

reinhard-brandstaedter, 2025-01-11:
Hab ich was falsch verstanden? Dachte da sf läuft bei dir als HA addon. Wenn nicht dann kann das HA schon laufen…nur halt keine automations die irgendwas mit dem hub oder sfcontrol machen.

reinhard-brandstaedter, 2025-01-11:
> joe@joe-ESPRIMO-Q920:/usr/local/sbin/sfcontrol$ sudo nano config.ini
> joe@joe-ESPRIMO-Q920:**/usr/local/sbin/sfcontrol$ sudo docker run -v ${PWD}/src/config.ini:/solarflow/config.ini --restart=always --name solarflow-control ghcr.io/reinhard-brandstaedter/solarflow-control:master**
> 2025-01-11 16:37:19,885:ERROR: No configuration file (config.ini) found in execution directory! Using environment variables.
> Traceback (most recent call last):
> File "/solarflow/solarflow-control.py", line 82, in
> or int(os.environ.get('MAX_INVERTER_INPUT',MAX_INVERTER_LIMIT - MIN_CHARGE_POWER))

Dein docker command mounted scheinbar keine config.ini weil diese im lokalen directroy nicht vorhanden ist?
Im docker container muss die config.ini auf jeden fall im /solarflow/ directory liegen:
```
% kubectl -n solarflow exec -it solarflow-control-5b7f74469b-jcfqm -- /bin/sh -c "pwd;ls -1"
/solarflow
__pycache__
config.ini
dtus.py
homeassistant
requirements.txt
smartmeters.py
solarflow-control.py
solarflow.py
utils.py
```

Btw. du verwendest nicht das aktuelle master release sondern vmtl. eines im cache. Dieser Fehler (TypeError: unsupported operand type(s) for -: 'int' and 'NoneType') kommt nicht beim aktuellen image.

reinhard-brandstaedter, 2025-01-11:
Kann es sein dass du irgendwann einmal manuell den battery_low wert auf 14% gestellt hast während der sf-control eigentlich in der discharge phase des CT modus war?
Nach den logs versucht er naemlich immer noch zu dischargen auf 0 runter, aber da der hub ein limit von low 14% gesetzt hat kann er den CT Zyklus nicht vollenden (der hub stoppt discharge bei 13% und erreicht nie 0%).

Irgendwo in den alten logs muss beim erreichen von 100% stehen dass er dann in die discharge phase geht und dabei setzt er das low-limit auf 0% und beim erreichen von 0 dann wieder auf das konfigurierte x% fuer den normalen ladezyklus.

joe63, 2025-01-12:
Hallo,
Ja ich gebe zu, dass ich einige Eingriffe gemacht habe da ich den CT-Zyklus anscheinend nicht richtig verstanden habe. Irgendwann nach dem restart des Containers habe ich den 0%-minimal-charge-wert im HA überschrieben, da ich nicht verstanden habe, dass der ct-zyklus eben so funktioniert.

Gibt es eine deutsche Doku zu deinem Projekt?

closed 2025-07-23 by reinhard-brandstaedter