reinhard-brandstaedter/solarflow-control #343: Min Charge Power - Battery is always charging

ralle6677, 2025-01-28:
Hi,

maybe I have a mistake in thinking. If the min charge power value is set to 125W, everything below 125W should go directly into the house and not into the battery. Am I thinking correctly? In my system, anything > 0W is always fed into the battery.

Maybe I chose the wrong setting. Having only been using Solarflow-control for a few days, I am still a beginner. :-)

Here is my Log:

2025-01-28 14:20:01,360:INFO: Location: Nuremberg, Bavaria, Germany
2025-01-28 14:20:01,360:INFO: Coordinates: (Lat: 49.4535, Lng: 11.1029
2025-01-28 14:20:01,360:INFO: No config setting found for option "disable_full_discharge" in section solarflow!
2025-01-28 14:20:01,360:INFO: No config setting found for option "rapid_change_diff" in section shellyem3!
2025-01-28 14:20:01,360:INFO: No config setting found for option "zero_offset" in section shellyem3!
2025-01-28 14:20:01,361:INFO: SF Control subscribing: solarflow-hub/KE30KTHh/control/#
2025-01-28 14:20:01,361:INFO: Reading retained config settings from MQTT...
2025-01-28 14:20:01,361:INFO: Note: Solarflow Control persists initial configuration settings in your MQTT broker and will use those first (if found) to allow on-the-fly updates!
2025-01-28 14:20:01,361:INFO: If you want to override these values from your config.ini you need to clear those retained topics in your broker first!
2025-01-28 14:20:01,447:INFO: Connected to MQTT Broker!
2025-01-28 14:20:01,451:INFO: Found control/minChargePower, set MIN_CHARGE_POWER to 125W
2025-01-28 14:20:01,452:INFO: Found control/batteryTargetSoCMin, set BATTERY_LOW to 30%
2025-01-28 14:20:01,452:INFO: Found control/dischargeDuringDaytime, set DISCHARGE_DURING_DAYTIME to False
2025-01-28 14:20:01,452:INFO: Found control/sunriseOffset, set SUNRISE_OFFSET to 60 minutes
2025-01-28 14:20:01,452:INFO: Found control/sunsetOffset, set SUNSET_OFFSET to 60 minutes
2025-01-28 14:20:01,452:INFO: Found control/batteryTargetSoCMax, set BATTERY_HIGH to 90%
2025-01-28 14:20:01,452:INFO: Found control/maxDiscahrgePiwer, set MAX_DISCHARGE_POWER to 150W
2025-01-28 14:20:11,362:INFO: Control Parameters:
2025-01-28 14:20:11,362:INFO:   MIN_CHARGE_POWER = 125
2025-01-28 14:20:11,362:INFO:   MAX_DISCHARGE_LEVEL = 150
2025-01-28 14:20:11,362:INFO:   MAX_INVERTER_LIMIT = 475
2025-01-28 14:20:11,362:INFO:   MAX_INVERTER_INPUT = 800
2025-01-28 14:20:11,362:INFO:   SUNRISE_OFFSET = 60
2025-01-28 14:20:11,362:INFO:   SUNSET_OFFSET = 60
2025-01-28 14:20:11,362:INFO:   BATTERY_LOW = 30
2025-01-28 14:20:11,362:INFO:   BATTERY_HIGH = 90
2025-01-28 14:20:11,362:INFO:   BATTERY_DISCHARGE_START = 10
2025-01-28 14:20:11,362:INFO:   DISCHARGE_DURING_DAYTIME = False
2025-01-28 14:20:11,364:INFO: Publishing Homeassistant templates...
2025-01-28 14:20:11,413:INFO: Published 36 Homeassistant templates.
2025-01-28 14:20:11,413:INFO: Triggering telemetry update: iot/73bkTV/KE30KTHh/properties/read
2025-01-28 14:20:11,414:INFO: Using OpenDTU: Base topic: solar/1164a00a9602, Limit topic: solar/1164a00a9602/cmd/limit_nonpersistent_absolute, SF Channels: [2], AC Limit: 475
2025-01-28 14:20:11,414:INFO: Using ShellyEM3: Base topic: shellies/shellyem3-485519DC66D1
2025-01-28 14:20:11,415:INFO: Hub subscribing: /73bkTV/KE30KTHh/properties/report
2025-01-28 14:20:11,415:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/solarInputPower
2025-01-28 14:20:11,415:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/electricLevel
2025-01-28 14:20:11,416:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/outputPackPower
2025-01-28 14:20:11,416:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/packInputPower
2025-01-28 14:20:11,416:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/outputHomePower
2025-01-28 14:20:11,417:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/outputLimit
2025-01-28 14:20:11,417:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/inverseMaxPower
2025-01-28 14:20:11,418:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/masterSoftVersion
2025-01-28 14:20:11,418:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/pass
2025-01-28 14:20:11,419:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/passMode
2025-01-28 14:20:11,419:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/socSet
2025-01-28 14:20:11,419:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/minSoc
2025-01-28 14:20:11,420:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/batteries/+/socLevel
2025-01-28 14:20:11,420:INFO: Hub subscribing: solarflow-hub/KE30KTHh/telemetry/batteries/+/totalVol
2025-01-28 14:20:11,420:INFO: Hub subscribing: solarflow-hub/KE30KTHh/control/#
2025-01-28 14:20:11,421:INFO: DTU subscribing: solar/1164a00a9602/0/powerdc
2025-01-28 14:20:11,421:INFO: DTU subscribing: solar/1164a00a9602/0/efficiency
2025-01-28 14:20:11,421:INFO: DTU subscribing: solar/1164a00a9602/+/power
2025-01-28 14:20:11,422:INFO: DTU subscribing: solar/1164a00a9602/status/producing
2025-01-28 14:20:11,422:INFO: DTU subscribing: solar/1164a00a9602/status/reachable
2025-01-28 14:20:11,422:INFO: DTU subscribing: solar/1164a00a9602/status/limit_absolute
2025-01-28 14:20:11,423:INFO: DTU subscribing: solar/1164a00a9602/status/limit_relative
2025-01-28 14:20:11,423:INFO: DTU subscribing: solarflow-hub/+/control/dryRun
2025-01-28 14:20:11,424:INFO: Shelly3EM subscribing: shellies/shellyem3-485519DC66D1/emeter/0/power
2025-01-28 14:20:11,424:INFO: Shelly3EM subscribing: shellies/shellyem3-485519DC66D1/emeter/1/power
2025-01-28 14:20:11,424:INFO: Shelly3EM subscribing: shellies/shellyem3-485519DC66D1/emeter/2/power
2025-01-28 14:20:11,425:INFO: Setting maximum charge level to 90%
2025-01-28 14:20:11,425:INFO: Setting minimum charge level to 30%
[... 29 lines cut ...]
2025-01-28 14:23:11,364:INFO: Triggering telemetry update: iot/73bkTV/KE30KTHh/properties/read
2025-01-28 14:24:11,365:INFO: Triggering telemetry update: iot/73bkTV/KE30KTHh/properties/read
2025-01-28 14:24:11,415:INFO: HUB: S:51.8W [ 51.8,51.8 ], B: 31% (31|31) low: 30 high: 90, V:49.4V (49.4|49.4), C: 47W, P:False (manual, possible), F:70.4h, E:220.9h, CT:OFF (120hrs) idle, H: 0W, L: 0W
2025-01-28 14:24:11,416:INFO: INV: AC:44.9W, DC:47.2W (0.0|0.0|23.9|23.2), L:211W (52.8W/channel) [1600W]
2025-01-28 14:24:11,416:INFO: SMT: T:ShellyEM3 P:213.9W [ 222.6,222.6 ]
2025-01-28 14:24:11,416:INFO: Direct connected panels (44.8W) can't cover demand (267.4W), trying to get 222.6W from hub.
2025-01-28 14:24:11,416:INFO: The current max direct channel power 22.7W is close to the current channel limit 52.8W, trying to get more from direct panels.
2025-01-28 14:24:11,416:INFO: Based on time, solarpower (51.8W) minimum charge power (125W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2025-01-28 14:24:11,416:INFO: Adjusting inverter limit from 22.7W to 53.1W
2025-01-28 14:24:11,416:INFO: Not setting inverter output limit as it is identical to current limit!
2025-01-28 14:24:11,416:INFO: Sun: 07:52 - 17:04 Demand: 267.4W, Panel DC: (44.8W), Hub DC: (0.0W), Inverter Limit: 212.0W, Hub Limit: 0.0W
2025-01-28 14:25:11,365:INFO: Triggering telemetry update: iot/73bkTV/KE30KTHh/properties/read
2025-01-28 14:26:11,365:INFO: Triggering telemetry update: iot/73bkTV/KE30KTHh/properties/read
2025-01-28 14:26:11,416:INFO: HUB: S:48.8W [ 48.8,48.8 ], B: 31% (31|31) low: 30 high: 90, V:49.5V (49.5|49.5), C: 44W, P:False (manual, possible), F:70.4h, E:220.9h, CT:OFF (120hrs) idle, H: 0W, L: 0W
2025-01-28 14:26:11,416:INFO: INV: AC:44.0W, DC:46.3W (0.0|0.0|23.5|22.8), L:211W (52.8W/channel) [1600W]
2025-01-28 14:26:11,416:INFO: SMT: T:ShellyEM3 P:214.5W [ 217.9,217.9,217.8 ]
2025-01-28 14:26:11,416:INFO: Direct connected panels (44.0W) can't cover demand (261.8W), trying to get 217.8W from hub.
2025-01-28 14:26:11,416:INFO: The current max direct channel power 22.3W is close to the current channel limit 52.8W, trying to get more from direct panels.
2025-01-28 14:26:11,417:INFO: Based on time, solarpower (48.8W) minimum charge power (125W) and bypass state (False), hub could contribute  0.0W - Decision path: 2.2.
2025-01-28 14:26:11,417:INFO: Adjusting inverter limit from 22.3W to 52.3W
2025-01-28 14:26:11,417:INFO: Not setting inverter output limit as it is identical to current limit!
2025-01-28 14:26:11,418:INFO: Sun: 07:52 - 17:04 Demand: 261.8W, Panel DC: (44.0W), Hub DC: (0.0W), Inverter Limit: 208.0W, Hub Limit: 0.0W

Thanks for your support!
Ralf

reinhard-brandstaedter, 2025-01-28:
Min Charge Power means that this is the value that - if available - should always be used for charging.
Think about it like this: the first 125W you are producing will be used to charge the battery.

ralle6677, 2025-01-28:
OK - that means the first 125W goes into the battery and the rest should go into the house? 

But with my system everything still goes into the house. There is no sun today, but there was some sun yesterday. I could observe that everything always goes into the battery.

reinhard-brandstaedter, 2025-01-28:
OK a addon-usecase.
Your setup sounds just right, identical to what I run (except a HMS-2000).

This defines the inverter's channel the hub is connected to, so that should be OK. (not too sure about the "", though a simple integer without " should also work, but maybe that is different in how the HA addon exports that to the config.

> sf_inverter_channels: "2"

From the log above everything seems to work fine, except that sf-control detected a 1600W maximum for the inverter?
The SF Panels are charging the battery and the direct connected panels are feeding to home.
A log with more sunshine, where the produced energy is above the min charge limit would maybe help.

Your Battery low limit is set to 30%, is that intentional?

ralle6677, 2025-01-29:
> This defines the inverter's channel the hub is connected to, so that should be OK. (not too sure about the "", though a simple integer without " should also work, but maybe that is different in how the HA addon exports that to the config.
> 
> > sf_inverter_channels: "2"

The quotes are only in the code. In the UI the value is recorded without "". I can't save it any other way in the code. He only saves the code with "". 

I hope there is more sun today or the next few days, then I will post a new log.

If you have a similar setup - how did you connect your hub to the inverter? First I split the output from the hub with a y-cable and went to inputs 1 and 2. But the hub sometimes reacted strangely. So the bypass didn't switch to "on" when the battery reached 100%. I can't say yet whether changing the cables made any difference. Too little sun. Maybe the hub has a hit?

Thanks for your help!

Gruß
Ralf

reinhard-brandstaedter, 2025-01-29:
NOT using the y-cable is the proper soluion.
The HMS has 4 MPPT trackers. When there is no galvanic separation (which is the case with a y cable) then the 2 MPPTs will “fight” for the best power point and influence eachother. This can lead to strange fluctuations an very bad performance. Typically you can then see up/downs on a V/A diagram of the inverter.
The HMS 1800 can handle ~700W per input so you should be fine when the hub feeds in up to 700W.

reinhard-brandstaedter, 2025-01-31:
@ralle6677 I guess you should have seen some correct charging in the past days?

ralle6677, 2025-01-31:
Servus Reinhard,

yes - today is quite a sunny day. The system charged to 100% through charge trough mode and set the bypass to on. So it could really have been the y-cable. However, I had to restart the Proxmox - after that the bypass switched to off again. I had to change it to on manually. 

Hopefully I'll see tomorrow how the system behaves after the battery is completely discharged. But I hope everything is working now. Will give feedback tomorrow.

Another question for you: Does solarflow control also take over the task of the openDTU's dynamic power limiter? Should I then deactivate the dpl in openDTU? Or does that not matter?

Thanks!!!!

reinhard-brandstaedter, 2025-02-01:
If you are using OpenDTU on battery with DPL you have potential conflicts as sf-control also limits the inverter. This will likely not work.
sf-control will limit to zero-feed in from battery but will not limit to zero from direct connected panels.

ralle6677, 2025-02-01:
Is openDTU without the battery option working? Or how did you fix that?

ralle6677, 2025-02-03:
I switched to OperDTU without Battery. Now it works to charge the battery with the set value. It's probably a combination of cabling with the y-cable and the OpenDTU on Battery.

All that's left now is for the bypass state to be changed when the target value has been reached. I haven't been able to test that yet. But I'm confident. 

Thank you for your support and your great solution!!! 💯

closed 2025-02-04 by ralle6677