reinhard-brandstaedter/solarflow-control #344: sf-control crashing when takinig action

PL-Peter, 2025-02-02:
I've got a SF Hub 1200 with 2 panels connected to the hub. The hub is hooked up with Y-cables to a Hoymiles inverter. There are no direct attached panels. 
Whenever sf-control decides it should take action, it crashes with the error below.

Firmware of the hub 1200 is latest as of this post, aswell as sf-control image being freshly pulled.

`sf-control  | 2025-02-02 10:27:09,207:INFO: HUB: S:37.4W [ 38.0,41.0,44.0,48.0,50.0,31.0,31.0,31.0 ], B: 18% (18), C: -35W, F:-1.0h, E:-1.0h, H: 65W, L: 75W
sf-control  | 2025-02-02 10:27:09,207:INFO: INV: AC:75.2W, DC:79.0W (), L: 0W
sf-control  | 2025-02-02 10:27:09,207:INFO: SMT: T:Smartmeter P:452.0W [ 508.0,438.0 ]
sf-control  | 2025-02-02 10:27:09,208:INFO: Direct connected panels can't cover demand  0.0W/527.2W, trying to get rest (527.2W) from SF.
sf-control  | 2025-02-02 10:27:09,208:INFO: Checking if Solarflow is willing to contribute 527.2W!
sf-control  | 2025-02-02 10:27:09,209:INFO: Sun: 07:21 - 16:56 - Solarflow limit:  0.0W - Decision path: 2.2.
sf-control  | Traceback (most recent call last):
sf-control  |   File "/solarflow/solarflow-control.py", line 446, in <module>
sf-control  |     main(sys.argv[1:])
sf-control  |   File "/solarflow/solarflow-control.py", line 443, in main
sf-control  |     run()
sf-control  |   File "/solarflow/solarflow-control.py", line 378, in run
sf-control  |     limitHomeInput(client)
sf-control  |   File "/solarflow/solarflow-control.py", line 335, in limitHomeInput
sf-control  |     inv_limit = inv.setLimit(max(remainder,getDirectPanelLimit(inv,hub,smt)))
sf-control  |   File "/solarflow/solarflow-control.py", line 176, in getDirectPanelLimit
sf-control  |     return math.ceil(max(inv.getDirectDCPowerValues())*rise_factor)
sf-control  | ValueError: max() arg is an empty sequence
sf-control exited with code 1
`

The configuration I'm running is

`[global]
dtu_type = AhoyDTU
smartmeter_type = Smartmeter

[solarflow]
device_id = [private]
full_charge_interval = 120
control_bypass = true
control_soc = true

[mqtt]
mqtt_host = 192.168.226.201

[ahoydtu]
base_topic = inverter
inverter_id = 0
sf_inverter_channels = [1,2]
inverter_max_power = 800
inverter_name = Home

[smartmeter]
base_topic = tele/tasmota_[private]/SENSOR
cur_accessor = E450.P_1
total_accessor = E450.E_1
rapid_change_diff = 500
zero_offset = 20

[control]
min_charge_power = 125
max_discharge_power = 500
max_inverter_limit = 800
max_inverter_input = 500

limit_inverter = true
inverter_min_limit = 10

discharge_during_daytime = false
battery_low = 10
battery_high = 99
`

It seems inv.getDirectDCPowerValues() might return an empty sequence in my setup, which causes the crash. Shouldn’t this case be handled gracefully?

reinhard-brandstaedter, 2025-02-04:
Thanks for reporting this. It seems to be related to how AhoyDTU is reporting channels. I’ll take a closer look. I’ve stopped using Ahoy in favor of OpenDTU.
I’ll take a closer look.

PL-Peter, 2025-02-06:
ChatGPT suggested to change line 176 from:
`return math.ceil(max(inv.getDirectDCPowerValues())*rise_factor)`

to:
`values = inv.getDirectDCPowerValues()
        if not values:
            return 0  # Or some default value that makes sense for your case
        return math.ceil(max(values) * rise_factor)`

This kept solarflow-control.py from crashing.

Is there a technical reason to prefer OpenDTU over AhoyDTU? I'd consider to switch ...