reinhard-brandstaedter/solarflow-control #348: Path 2.1.2 could be improved

BuddhaOhneHals, 2025-02-17:
Hi,

I had the following scenario today:

- Solar power was something like 1 kW
- Demand was more than solar power (about 1.8 kW)
- Output to home (Inverter Limit) was set to 300 W

I think by having DISCHARGE_DURING_DAYTIME set to True, the Inverter Limit is set to MAX_DISCHARGE_POWER (which is set to 300 W), [here at line 300](https://github.com/reinhard-brandstaedter/solarflow-control/blob/master/src/solarflow/solarflow-control.py#L300C40-L300C59).

I would expect the Inverter limit set to `min(demand, <Solar power> + MAX_DISCHARGE_POWER)`. Or am I missing something?

Some logs:
```
2025-02-17 08:46:51,842:INFO: HUB: S:936.7W [ 933.1,933.1,933.0,932.9,933.5,936.7 ], B: 35% (35) low: 15 high: 100, V:51.0V (51.0), C: 591W, P:False (auto, possible), F:18.9h, E:-1.0h, CT:OFF (96hrs) idle, H:319W, L:800W
2025-02-17 08:46:51,842:INFO: INV: AC:309.0W, DC:326.2W (159.0|167.4), L:300W (150.0W/channel) [800W]
2025-02-17 08:46:51,842:INFO: SMT: T:Smartmeter P:1836.0W [ 1879.0,1879.0,1838.7,1841.0 ]
2025-02-17 08:46:51,843:INFO: Direct connected panel are producing 0.0W, trying to get 2150.2W from hub.
2025-02-17 08:46:51,844:INFO: Based on time, solarpower (936.7W) minimum charge power (100W) and bypass state (False), hub could contribute 300.0W - Decision path: 2.1.2.
2025-02-17 08:46:51,844:INFO: Not setting solarflow output limit to 800.0W as it is identical to current limit!
2025-02-17 08:46:51,845:INFO: Solarflow is willing to contribute 150.0W (per channel) of the requested 2150.2!
2025-02-17 08:46:51,846:INFO: Not setting inverter output limit as it is identical to current limit!
2025-02-17 08:46:51,847:INFO: Sun: 07:42 - 17:52 Demand: 2150.2W, Panel DC: (0.0W), Hub DC: (309.2W), Inverter Limit: 300.0W, Hub Limit: 800.0W
```
Thanks,
Timo

reinhard-brandstaedter, 2025-02-20:
The assumption here is that, even throughout the day we do not want to use more than the MAX_DISCHARGE_POWER from the hub, so that a consitent charge of the battery is also possible.
Ideally you have panels not only connected to the hub but also to the inverter, which contribute to lowering the demand during the day.
If you want to use more from the hub (be it battery or excess sun) you could just raise MAX_DISCHARGE_POWER to the max of your inverter input limit but setting it to  ```<Solar power> + MAX_DISCHARGE_POWER``` doesn't make sense, you might fry your inverter or your fuses.
The excess is rather used to charge the battery and then covering long-term load down to zero than trying to cover max/peaks.

In the end you should tune the MAX_DISCHARGE_POWER / MIN_CHARGE_POWER in a way that the battery gets full during the day, without too long bypass times and gets completely emptied throughout the night.

tuxianerDE, 2025-05-08:
@BuddhaOhneHals closing the issue as it has been inactive. if you want to reopen you can.

closed 2025-05-08 by tuxianerDE