reinhard-brandstaedter/solarflow-control #349: ByPass not enabled in case battery is not depleted overnight

dfigus, 2025-03-07:
With the recent sunny and warm weather and not consuming much energy, I get through the night without depleting the battery. In this case I had it multiple times that solarflow-control does not enable the bypass after reaching 100% even thought everything is correctly configured (it also worked in the past with depleted battery).

I had a quick static analysis and it seems that in this case the `batteryTarget` never switches to something else than `discharging`. But this is checked when enabling the bypass:

https://github.com/reinhard-brandstaedter/solarflow-control/blob/2d9ed3484ec70b681c407645a13a0fba59762a35/src/solarflow/solarflow.py#L219-L227

In this case `batteryTarget` and `self.batteryTarget` are still `discharging`. Also I could not find any place where `self.batteryTarget` is reset apart from handling 0% or any given user specific min value:
https://github.com/reinhard-brandstaedter/solarflow-control/blob/2d9ed3484ec70b681c407645a13a0fba59762a35/src/solarflow/solarflow.py#L200-L217

I gave it a quick thought bit didn't find a good solution so far. Maybe you have a good idea. I keep thinking of a solution as well and raise a PR in case I have one.

DB6CP, 2025-04-09:
I can reproduce this behavior. I am not running empty anymore due to more and longer Sunlight. I have to switch Bypass Manually at the moment.

dfigus, 2025-04-10:
@DB6CP You can apply the changes from my PR #350 (e.g. via `patch`). It's running on my end for around 1 month without issues.

DB6CP, 2025-04-24:
Did the Update yesterday and updated the Config just now. As soon as the weather allows it i will Report if it works.

DB6CP, 2025-05-07:
Seems to work finde. No Problems so far.

tuxianerDE, 2025-05-14:
hey folks,

I will review PR #350 and see if we accept the changes that have accumulated there around DTU, SF and SmartMeter.

reinhard-brandstaedter, 2025-05-19:
Sorry for not chiming in earlier. I was on a 2 months leave far, far away enjoying sun and beaches :-)
I definitely can confirm this behavior and will take a look/merge the fix.