reinhard-brandstaedter/solarflow-control #355: Firmware 2.0 Hoymiles (Limiter)

busaku, 2025-03-28:
Hey Guys,

the new Firmware 2.0 for the Hoymiles Inverter works a little bit different from the 1.* Versions.
Setting the inverter limit = setting the output limit instead of input / (mppt count) for the different MPPTs.

Is there a easy way to prevent setting / using inverter limiting?

reinhard-brandstaedter, 2025-04-01:
I don’t think thats a differen behavior.
The limit is always set on the output, but its always assumed that every mppt contributes equally to the limit.
Do you have any details from hoymiles that this changed in 2.x?

reinhard-brandstaedter, 2025-04-06:
Thanks for the link!
So if I’m correct then starting with 1.01.12 this new power distribution logic is in place.
Actually a fix for a long time issue that everyone needed to work around.
But for a even newer FW ther won’t be sny control at all.

So to compensate I’d need to add a FW check and calculate differently.

Will see to that once I have some time again.