reinhard-brandstaedter/solarflow-control #357: Parameters in smartmeter-section for Hichi

marc107, 2025-04-02:
Hello Reinhard,

thank you very much for publishing this great project!
Regarding the use of a Hichi, I have one more question before I order the Hyper 2000 and get a rude awakening ;-)

At the moment I am already reading the total power consumption via the Hichi and publishing it in a mqtt topic via fhem.

The config.ini contains the following:

```
> [smartmeter]
> # The MQTT base topic your Hichi, Tasmota, generic smartmeter reader reports to
> base_topic = tele/E220/SENSOR
> # if the published value at the base_topic is a JSON type then these accessors are used to get the power values
> # e.g. if Smartmeter reader posts { “Power”: {“Power_curr”: 120, “Total_in”: 12345.6} }
> cur_accessor = Power.Power_curr
> total_accessor = Power.Total_in
```

As I understand it, the power consumption value of my Hichi corresponds to “total_accessor = Power.Total_in”, which is important for zero power control. Is this value alone  sufficient for pure AC charging via Hypper 2000 or is the current (cur_accessor) also required?

Best regards

Marc

reinhard-brandstaedter, 2025-04-06:
cur_accessor is the current power consumption in W.
This is what is required, the total_accessor is not required.

marc107, 2025-04-07:
Thank u very much!

closed 2025-05-08 by tuxianerDE