reinhard-brandstaedter/solarflow-control #358: Bei 100% Akku wird der Eingang abgeschaltet

sturowski, 2025-04-27:
Hallo, 
ich habe mal wieder ein Problem. Ist aber bestimmt nur Konfiguration. 
Ich habe gerade festgestellt, das sobald der Akku auf 100% ist, dann scheint der Eingangskanal abgeschaltet zu werden.

Hier ein Screenshot:

[image]

Ich hoffe du kannst mir helfen. 
Liebe Grüße
Sven

sturowski, 2025-04-27:
Der Akku warum um 17:30Uhr voll. Dann ging relativ schnell die Leistung runter auf 0w, trotz das die Sonne weiter schien. Um 18:00Uhr ging die Leistung wieder hoch. Bei genau 30min sagt mir mein Entwicklergehirn, dass ist kein Zufall.

Hast du etwas eingebaut in die Richtung oder kommt das von Zendure selber?

Eine Möglichkeit wäre, wenn der Akku bei 99% oder sagen wir 95% ist, dass laden zu unterbinden und nur alle x Tage den dann auf 100% zu schicken. Weil das waren jetzt bestimmt 500w über 30min, die weg waren.

tuxianerDE, 2025-05-08:
Hey, 

welchen Wert hast du denn in der Konfig für das bypass Handling gesetzt bzw. im Hub?

Hintergrund:
Wenn die Akkus voll sind (generell ist das die Charge Grenze also kann auch 99% je nachdem was du eingestellt hast), werden die Panele runtergeregelt bis der bypass aktiviert wird. Dann gehen die Panele direkt auf den WR durch. ABER, wenn byPass auf OFF steht (also der Hub darf ihn nicht aktivieren oder das Control skript) dann regelt er den die Eingänge runter und tut nichts weiter.

2te Möglichkeit (Bug/Verhalten aktuell): Wenn du im CT Modus bist (also ChargeThrough aus dem sf-control) will das Skript einmal auf 100% und komplett runter auf 0%. Erst danach wird der CT Modus abgeschaltet bis das interval wieder erreicht ist. Wenn die Batterie aber nicht geleert wird (also 0% nicht erreicht), dann beginnt er diese wieder zu laden und schaltet auch dort am Ende bei 100% nicht in den byPass.

sturowski, 2025-05-09:
Hallo,
ich habe Bypass auf false und im HUB selber abgeschaltet, also kein ByPass möglich. Ich hatte jetzt mehrere Tage immer wieder 100% und nach 30min geht alles wieder. Ist zwar komisch, aber denke das liegt am Verhalten des HUBs.

sturowski, 2025-05-09:
Kannst du mir sagen, warum ich hier nicht mehr die Parameter anpassen kann:
# flag, which can be set to allow discharging the battery during daytime
# MQTT config topic: solarflow-hub/control/dischargeDuringDaytime
# config.ini [control] discharge_during_daytime
DISCHARGE_DURING_DAYTIME = None

#Adjustments possible to sunrise and sunset offset
# MQTT config topic: solarflow-hub/control/sunriseOffset
# config.ini [control] sunrise_offset
SUNRISE_OFFSET = None
# MQTT config topic: solarflow-hub/control/sunsetOffset
# config.ini [control] sunset_offset
SUNSET_OFFSET = None

Warum wurde das aus der config.ini entfernt? Oder finde ich nur die Stelle nicht, wo das gesetzt wird?

tuxianerDE, 2025-05-09:
Wie kommst du drauf, dass du diese Werte nicht mehr anpassen kannst?  Gerade in meiner ini geschaut, die Werte stehen bei mir in der [control] (also die offsets).

Was du ggf. beachten musst, wenn du die Werte änderst, musst du auch den control Teil in deiner MQTT löschen. Dort werden die Werte persistiert und du willst sie natürlich überschreiben.

sturowski, 2025-05-12:
Dann habe ich den Code falsch gelesen. Wo wird das denn eingelesen? Dachte immer, das geht hiermit 
`config.getint` und das finde ich dafür nicht.

Kann ich nicht einfach DISCHARGE_DURING_DAYTIME auf True setzen? Dann müsste der aus den Akku nehmen, sobald er es braucht korrekt? Also auch am Tag. Bin aktuell bei Sonnenaufgang noch bei 40-50% Akku und ich kaufe Strom ein, weil die Sonne noch nicht reicht. 

Oder hat das irgendwelche Nachteile für den HUB oder den Akku?

tuxianerDE, 2025-05-13:
Guten Morgen,

Korrekt ich habe den Wert z.B. Auch immer auf true gesetzt so dass auch während des Tages sich am hub bzw den Batterien bedient. Einzige Ausnahme ist da der Charge Through Modus, da ist natürlich das Ziel so schnell es geht wieder auf 100% zu kommen.

sturowski, 2025-05-13:
Sehr cool, vielen Dank, dann werde ich das nachher mal umstellen bei mir 👍 
Bist du schon lange in dem Projekt? Ich verwende die Software jetzt seit Juni letzten Jahres und einfach nur mega cool.

tuxianerDE, 2025-05-13:
Die Software ist bei mir seit gut 1 1/2 Jahren im Einsatz bzw nehme die Entwicklungen mit. Ich glaube (und @reinhard-brandstaedter ) müsste es korrigieren ich glaube ich war damals der erste PR und hatte die MQTT Authentifizierung eingebaut :D :D bin aber kein Hardcore Developer

sturowski, 2025-05-13:
Ahh cool :-) zu mir als Hintergrund, ich bin Seit knapp 15 Jahren beruflich Softwareentwickler. Seit 10 Jahren programmiere ich hochverfügbare kritisch Systeme. Alles in C++. Daher finde ich das Projekt auch so cool. Seit diesem Jahr bin ich der Leiter der Softwareentwicklung in meiner Firmengruppe.
Ich habe jetzt discharge_during_daytime auf true gesetzt, mal gucken wie das ist. Hier ist meine aktuelle control Config:
```
[control]
min_charge_power = 120
max_discharge_power = 200
max_inverter_limit = 800                                                
limit_inverter = true
inverter_min_limit = 10
battery_low = 20
battery_high = 100
max_inverter_input = 700
discharge_during_daytime = true
```

Ich bin am überlegen, ob ich max_discharge_power höher setzen soll, bzw battery_high auf 98% oder so, damit der nie in diese 100% 30min Abschaltung geht. Hast du damit erfahrungen?

tuxianerDE, 2025-05-14:
Ich habe battery high und battery low mit 99% und 1%. Den SoC Werten kannste nicht trauen.

sturowski, 2025-05-14:
Wie sieht es bei dir aus mit max_discharge_power ?

sturowski, 2025-05-15:
danke :-)

closed 2025-05-17 by tuxianerDE