reinhard-brandstaedter/solarflow-control #360: solarflow-control subscribes to wrong topic

JM-HH, 2025-05-05:
Hello,
I have a Hyper 2000 disconnected from the cloud. It writes the data to the topic /gDa3tb/xxxxxxxx on mosquitto.
Mosquitto and solarflow-control are running on Docker.
In the config.ini of solarflow-control I have defined the product_id
`[solarflow]

product_id = gDa3tb
device_id = xxxxxxxx
`
The log shows at startup
2025-05-05 17:41:04,310:INFO: Solarflow Hub: gDa3tb/xxxxxxxx
and
025-05-05 17:41:04,489:INFO: Connected to MQTT Broker!
2025-05-05 17:41:04,489:INFO: Hub subscribing: /**73bkTV**/xxxxxxxx/properties/report

solarflow-control subscribes to the wrong topic, any hints?

reinhard-brandstaedter, 2025-05-05:
Can you check if your config.ini is properly mounted into the Docker container?

JM-HH, 2025-05-06:
Hello Reinhard,

I added a dummy entry to the config.ini and restarted the container, the dummy entry is visible from inside the container. So it should be properly monted.

JM-HH, 2025-05-07:
Hello Reinhard,

in the docker-compose.yml in examples/local_mqtt_with_control/docker-compose.yml the image rbrandstaedter/solarflow-control:**refactoring** is used. In this image the product_id is hardcoded. With the latest image it runs.

Regards
Jochen

tuxianerDE, 2025-05-08:
@JM-HH that is of course not correct in the example. Are you using the regular master now and was that the problem that you used the refactoring image?

reinhard-brandstaedter, 2025-05-19:
@JM-HH you should use the latest image from [here](https://github.com/reinhard-brandstaedter/solarflow-control/pkgs/container/solarflow-control/418768743?tag=master)